Makuna/NeoPixelBus #510: Cannot delete and re-create NeoPixelBus on esp32

jackjansen, 2021-08-07:
**Describe the bug**
This bug is related to #501, but #504 doesn't solve it completely.

> My use case is that I have a neopixel strip with a MOSFET to disconnect GND (driven by a GPIO pin). This way, when I know the whole strip is off I can disconnect it from power (which saves quite a bit because each NeoPixel control chip uses about 1mA in rest). But I also have to float the DATA pin, otherwise leak current will dimly light the first LED.

The `gpio_matrix_out` from #504 now floats the DATA pin if I `delete` the NeoPixelBus, so that is great.

But when a create a `new NeoPixelBus()` later because I want to turn the strip on again nothing happens.

I _think_ this is because `Esp32_i2s.c` uses some static structures and it thinks it doesn't have to do the `gpio_matrix_out()`, not being aware of the fact that the previous call has been undone. But I'm not 100% sure.

**To Reproduce**
- Create a `new NeoPixelBus()` that uses the esp32 i2s driver.
- Delete it.
- fiddle with the GIO DATA pin.
- Re-create a new NeoPixelBus with the same parameters, pins.
- Note that it doesn't work.

Makuna, 2021-08-08:
I just merged some cleanup with this pull https://github.com/Makuna/NeoPixelBus/pull/511; get the latest and give it a try.

Makuna, 2021-08-09:
This is really a duplicate of a previous issue https://github.com/Makuna/NeoPixelBus/issues/236

closed 2021-08-09 by Makuna

jackjansen, 2021-08-11:
I can confirm that the fix works, thanks!

(for reference for interested parties stumbling on this issue, https://github.com/cwi-dis/lissabon/tree/issue8-dimpixel simply cuts power to the led strip and deletes the NPB when all leds are off, and applies power and re-creates the NPB when any pixels are on. This now works, even with deep-sleep)