Makuna/NeoPixelBus #516: 16 Bit RGBW Support for UCS8904

Differentunic, 2021-10-06:
I have a couple hundred UCS8904 leds, which are RGBW. I've tried to get them to work with WLED, but the colors see random for the whole length of leds. 

I've been experimenting with getting [FastLED](https://github.com/FastLED/FastLED) to work with RGBW, but the same problem with random colors ocoured. I spent a couple hours translating a Chinese pdf (to the best of my ability) that I found online. I created a temp repo [here](https://github.com/Differentunic/UCS8904-Datasheet/blob/main/UCS8904B.pdf). This should have everything you need.

Regards,
Differentunic

Differentunic, 2021-10-06:
Yep, I'm able to test everything. It's 60 leds per strip, but when I did my FastLED trick I had to enter 2x the amount for some reason.

At the moment I have an Arduino leanardo, uno, mega, and esp2866 for testing.

Thanks

Differentunic, 2021-10-08:
We have success!!!

https://user-images.githubusercontent.com/66045103/136509920-e8656141-0127-4104-ad1d-fceb9c048e70.MOV

Thank you @Makuna for your hard work!

Makuna, 2021-10-08:
AND, not sure you noticed this, but these LEDs actually have a 16bit resolution for each color element; so you can use the `Rgb48Color::Max` or `Rgb64Color::Max` (which is 65535 instead of 255) and give your LEDs a far smoother gradients than WS2812x LEDs.

Makuna, 2021-10-11:
https://github.com/Makuna/NeoPixelBus/pull/517

NGomes7, 2022-05-12:
Hi, can anyone provide a brief overview on UCS8904 communication protocol please? I want to interface a MCU ESP8266 with an LED strip containing several UCS8904 but I am not sure how many bits I should send a time in order to encode the Colors RGBW. I know that, for WS2815B, I had to send 24 bits to encode the colors RGB. 
Thanks for the help!