Makuna/NeoPixelBus #524: Odd output corruption on multiple GPIO outputs using large LED count

blazoncek, 2021-10-13:
**Describe the bug**
We at [WLED](https://github.com/Aircoookie/WLED) are experiencing an odd issue when we define multiple outputs (different GPIO ports) with large LED count (>2000).
Currently a well documented issue is 75x40 LED matrix using 5 GPIO outputs (5x600=3000 pixels) on ESP32.
If the number of GPIOs is reduced (keeping the same total LED count) the output corruption is reduced to occasional flicker.
The output corruption can also be measured with logic analyser on actual GPIO pins.

**To Reproduce**
Steps to reproduce the behavior:
1. Create 75x40 WS2812B matrix using 5 x 600 LED strips
2. Use WLED 0.13-b4 and assign GPIOs 16, 3, 23, 26, and 27 with LED count of 600 each
3. Start any effect available in WLED
4. See corrupted output or occasional flicker

**Expected behavior**
No output corruption

**Development environment (please complete the following information):**
 - Build Environment [VSC+PIO; platform=espressif32@2.0]
 - Board target [ESP32 WROOM]
 - Library version [v2.6.7]

**Minimal Sketch that reproduced the problem:**
Unfortunately N/A.

**Additional context**
Everything seems fine with low GPIO ports used and lower LED count.
Free heap is still plenty at 150kB.
Video of corruption available upon request.
Contact persons involved are also @jdavis7765 and @pbolduc with WLED issue [#2268](https://github.com/Aircoookie/WLED/issues/2268)

The issue seems to disappear when downgrading to WLED 0.12 which used NeoPixelBus 2.6.0.

embedded-creations, 2021-12-11:
I'm wondering if the I2S parallel output driver integrated into FastLED is worth porting over to NeoPixelBus.  For WLED, after 5 WS2812 outputs are enabled, it could disable the RMT drivers and start using I2S.  I'm not sure of the RAM cost to compare RMT and I2S.

https://www.reddit.com/r/FastLED/comments/bjq0sm/new_24way_parallel_driver_for_esp32/

Makuna, 2021-12-11:
@embedded-creations Try to keep on topic within "Issues".  See https://github.com/Makuna/NeoPixelBus/issues/270 for what you are suggesting.

blazoncek, 2022-01-24:
@Makuna do you agree to close this issue since it may be related to core RMT buffer behavior?

Makuna, 2022-01-24:
@blazoncek That makes the most sense at this point.  We can always reopen if new information appears.

closed 2022-01-24 by blazoncek