Makuna/NeoPixelBus #535: declaration of DotStarWbgrFeature missing?

beamzer, 2021-12-01:
Hi, i have a LPD8806 "DotStar" string with sequentially a RGB LED followed by a white LED. I used the DotStarTest.ino from the examples directory and uncommented (what is thought was) the right config for my setup:
```
// DotStars that support RGBW color with a separate white element
NeoPixelBus<DotStarWbgrFeature, DotStarMethod> strip(PixelCount, DotClockPin, DotDataPin);
```
But it seems that DotStarWbgrFeature is not declared anywhere? The software only shows it in the examples but not in the .h or .c files.

The error i get is:
```
DotStarTest:28:13: error: 'DotStarWbgrFeature' was not declared in this scope; did you mean 'DotStarLbgrFeature'?
   28 | NeoPixelBus<DotStarWbgrFeature, DotStarMethod> strip(PixelCount, DotClockPin, DotDataPin);
      |             ^~~~~~~~~~~~~~~~~~
      |             DotStarLbgrFeature
```

thanks in advance,
Ewald...

Makuna, 2021-12-01:
LPD8806 has its own features, listed in the `Lpd8806ColorFeatures.h`.  
Which are
* Lpd8806BrgFeature
* Lpd8806GrbFeature

Note: The Wiki still hasn't been updated with all the custom DotStar required color features yet.

As far as I can find, there are no RGBW variants of this chip (LPD8806 is the chip id).  Do you have a link to the product you purchased that is RGBW?

Makuna, 2021-12-03:
try this declaration...
```
NeoPixelBus<Lpd8806GrbFeature, Lpd8806Method> strip(PixelCount, DotClockPin, DotDataPin);
```
and then have it output these pixel colors...
```
// should be white LED only
strip.SetPixelColor(0, RgbColor(0,128,0));
strip.SetPixelColor(0, RgbColor(0,0,0));
// should be green
strip.SetPixelColor(1, RgbColor(128,0,0));
strip.SetPixelColor(1, RgbColor(0,0,0));
// should be red
strip.SetPixelColor(2, RgbColor(0,0,128));
strip.SetPixelColor(2, RgbColor(0,0,0));
// should be blue
strip.SetPixelColor(3, RgbColor(0,0,0));
strip.SetPixelColor(3, RgbColor(0,128,0));
// should be white using rgb
strip.SetPixelColor(4, RgbColor(128,0,128));
strip.SetPixelColor(4, RgbColor(0,128,0));
strip.Show();
```
you should see the first five pixels set to 50% brightness with this color order (white only led, green, red, blue, white using rgb leds)

beamzer, 2021-12-04:
Thanks, that helped me to get it working, really appreciate it.

Apparently my string is RBG with your code, but that doesn't matter. The white LEDs also have 3 LED's per pixel, but they are all white, so you can get quite a brightness when switching them all on, The strip starts with a color pixel.

Example of the working code for my Wemos D1 mini (esp8266):
`
#include <NeoPixelBus.h>

// Pins for Wemos D1 mini (ESP8622)
const uint8_t DotClockPin = D5; // SCLK (LPD8806 strip => green wire)
const uint8_t DotDataPin  = D7; // MOSI (LPD8806 strip => yellow wire)

const int dly = 1000;

NeoPixelBus<Lpd8806GrbFeature, Lpd8806Method> strip(PixelCount, DotClockPin, DotDataPin);

RgbColor red(128,0,0);
RgbColor green(0,0,128);
RgbColor blue(0,128,0);
RgbColor white(128,128,128);
RgbColor black(0,0,0);

RgbColor white1(128,0,0);
RgbColor white2(0,128,0);
RgbColor white3(0,0,128);

void setup()
{
    // this resets all the LEDs to an off state
    strip.Begin();
    strip.ClearTo(black);
    strip.Show();
}

void loop()
{

// Red
strip.SetPixelColor(0, RgbColor(red));
strip.SetPixelColor(1, RgbColor(white1));
strip.SetPixelColor(2, RgbColor(red));
strip.SetPixelColor(4, RgbColor(red));
strip.Show();
delay(dly);

// Blue
strip.SetPixelColor(0, RgbColor(blue));
strip.SetPixelColor(1, RgbColor(white2));
strip.SetPixelColor(2, RgbColor(blue));
strip.SetPixelColor(4, RgbColor(blue));
strip.Show();
delay(dly);

// Green
strip.SetPixelColor(0, RgbColor(green));
strip.SetPixelColor(1, RgbColor(white3));
strip.SetPixelColor(2, RgbColor(green));
strip.SetPixelColor(4, RgbColor(green));
strip.Show();
delay(dly);

// White
strip.SetPixelColor(0, RgbColor(white));
strip.SetPixelColor(1, RgbColor(white));
strip.Show();
delay(dly);

strip.ClearTo(black);
strip.Show();
delay(dly);

}
`

Makuna, 2021-12-04:
Tell me exactly what LED state you see when using my code.  I need this to understand how to expose the right feature to support it.  Until we get the feature correct, the actual colors you pass to `SetPixelColor` are not correct.

Your example skips pixel 3 for some reason?

If you are using the actual SPI pins on the ESP8266, then you can use the method `Lpd8806SpiMethod` for hardware SPI support (faster and asynchronous)

beamzer, 2021-12-05:
I omitted pixel three because i didn't want to draw to much current from the USB bus, it's easier testing without an external power-supply. But all the evens are colour on my string and the odds are white. They all contain three LEDs per pixel, but the white pixels are just all white. The ```Lpd8806SpiMethod```  is indeed noticeably faster, not that i need it in my application but it's the most efficient drive method anyway.
For initialisation i also switched the ```Lpd8806GrbFeature``` to ```Lpd8806BrgFeature``` and that way the values in ```RgbColor``` correspond to the R-G-B sequence again.
so it's now:
```NeoPixelBus<Lpd8806BrgFeature,Lpd8806SpiMethod> strip(PixelCount, DotClockPin, DotDataPin);```

Makuna, 2021-12-05:
Are you stating that the order of strip is BRGWxx or BRGWWW? Are each of the three whites addressable separately or does the first value in RgbColor get applied to all three?

try using 63 instead of 128 to lower the power usage when doing this sort of lower level work.  It also helps with looking directly at them ;-)

beamzer, 2021-12-06:
It's a RGBW strip, each of the three white LEDs in the odd numbered pixels are separately addressable, but they emit the same colour temperature. I bought it here https://nl.aliexpress.com/item/32475750463.html to be used for a wordclock, so the spacing is critical. After i found out that the white LEDs were not integrated with the colour LEDs i complained and got a refund of 50% ;). So i am going to use this for generic lighting somewhere else.

Makuna, 2021-12-06:
Ok, with this detail, then the best solution (as long as the current color order works, and you state it does) is to leave it to sketch author to understand odd pixels as just white. 

So this issue requires no more work and should be closed.

beamzer, 2021-12-06:
thanks for your support.

closed 2021-12-06 by beamzer