Makuna/NeoPixelBus #541: RgbwColor, Rgb48Color and Rgb64Color need Gamma correction support

Makuna, 2021-12-14:
**Is your feature request related to a problem? Please describe.**
Currently the Gamma correction only supports RgbColor

**Describe the solution you'd like**
Implementations to support these other BASE color objects used by Features and Methods

**Additional context**
For 16bit per color element, just adjusting the major byte is close enough.  This allows for faster conversion with least amount of memory for the table lookup versions.

MicePauls, 2021-12-28:
Hi,
Could you take us a sample for TM1814 RGBW, please.

Makuna, 2021-12-29:
https://github.com/Makuna/NeoPixelBus/pull/542

@MicePauls look at the provided library example NeoPixelGamma and use the NeoGammaEquationMethod otherwise you will lose too much color fidelity.

MicePauls, 2021-12-30:
@Makuna Thank for answer me.
I have a TM1814 strip led, and used this library to program it. But it has this error:

"TM1814_neobus:50:35: error: 'NeoTm1814Method' was not declared in this scope
 NeoPixelBus<NeoWrgbTm1814Feature, NeoTm1814Method> strip(PixelCount, PixelPin);"
I'm using Arduio IDE 1.8.13, Could you explain me about it, please?

Makuna, 2021-12-30:
@MicePauls Issues for bugs/feature tracking, not asking questions.  Further, it is considered poor form to use a tracking issue for discussing a problem outside the scope of the issue.
Ask your questions in the [discussions area](https://github.com/Makuna/NeoPixelBus/discussions) and I will answer it.

MicePauls, 2021-12-30:
Thank you so much.

Makuna, 2022-05-27:
v2.7.0

closed 2022-05-27 by Makuna