Makuna/NeoPixelBus #548: Make it possible to invoke ->Show() from e.g. inside a Ticker()-Callback. (yield-problem)

simon-budig, 2022-01-01:
I am experiencing weird crashes on an esp8266 based platform. I consider smoothness of my animation as one of the top priorities and network-libraries used in loop() cause all kinds of timing issues.

So I am rendering and Show()ing the animation from within a Ticker()-Object (I am using platformio with the Arduino Framework), where a callback is attached to be called every 16ms. This produces nice smooth animations.

*However* I sometimes experience crashes with quite weird stack traces and it seems to be related to calling yield() from inside a Ticker-Callback. Here is an issue in a different Library that seems kind of related: https://github.com/adafruit/Adafruit-GFX-Library/issues/285

NeoPixelBus calls yield() in many Update()-Functions, making it apparently unreliable to use in Ticker-Callbacks. It might be feasible to consider alternate ways to implement waiting.

simon-budig, 2022-01-01:
for reference here is the top of the stack trace where one of these kinds of crash is happening:

```
0x402076fd in render_worms(NeoPixelBus<NeoGrbwFeature, NeoEsp8266UartMethodBase<NeoEsp8266UartSpeed800Kbps, NeoEsp8266Uart<UartFeature1, NeoEsp8266UartContext>, NeoEsp8266UartNotInverted> >*, unsigned int, unsigned char, _renderfunc_param const*, unsigned int*, unsigned int*) at ??:?
0x4021549e in raise_exception at core_esp8266_postmortem.cpp:?
0x4021550b in __panic_func at ??:?
0x40214f61 in yield at ??:?
0x4020589c in _ZN11NeoPixelBusI14NeoGrbwFeature24NeoEsp8266UartMethodBaseI26NeoEsp8266UartSpeed800Kbps14NeoEsp8266UartI12UartFeature121NeoEsp8266UartContextE25NeoEsp8266UartNotInvertedEE4ShowEb$isra$0 at esp2ws.ino.cpp:?
0x4020722f in render_frame() at ??:?
0x40100421 in millis at ??:?
0x40227c57 in lwip_cyclic_timer at /home/earle/src/esp-quick-toolchain/arduino/tools/sdk/lwip2/builder/lwip2-src/src/core/timeouts.c:261
0x40105c49 in wdt_feed at ??:?

```

Makuna, 2022-01-01:
If you don't want the yield to be called, then you can check the state of the async sending by calling `CanShow()`, implement your own wait mechanism, so when you do call `Show()` it should not yield.

But note, in the past, yield wasn't the only issue, the actual calls to start the show (i2s DMA hardware calls) could not reliably be called within a ticker.  Which means this will not get fixed.

simon-budig, 2022-01-01:
Thank you, this is helpful, I will try to figure out if busy waiting before Show()ing is enough to drop the number of crashes to an acceptable level.

Is it just me or is the whole mess around the scheduling/"threading" stuff within the arduino framework woefully underdocumented?

Makuna, 2022-01-02:
Threading is not an Arduino standard at all and is a complex topic to understand.  Each implementation of it will have nuances to each platform.  Ticker is Esp8266 specific, isn't it?  ESP32 is on top of RTOS so you use its threading model (loop runs within a thread here).

I wrote a library (Task) that does non-preemptive switching, which is compatible with ANY platform but requires you to rethink how you write code and is fully compatible as everything runs within the loop context.  But this doesn't really make it easier to understand to use tasks/threads/fibers.  And being non-preemptive means if you write bad code, you can easily starve other tasks of CPU as there is no guaranteed call, the thread has to let others play.

simon-budig, 2022-01-02:
I use Ticker as described in e.g. https://circuits4you.com/2018/01/02/esp8266-timer-ticker-example/ - so yeah, this seems esp8266 specific. There also is an arduino Ticker-Library but this needs explicit invoking within the loop(), so that is something else and not suitable for my needs.

The problem for me is, that it seems that under some circumstances e.g. ArduinoOTA.handle() within loop takes an absurd amount of time (about a second or even more, to the point of causing the Software-Watchdog to trigger - wtf?) and it totally makes messes up all of the timing. So everything that is not preemptive will be messed up by it. So that means I would have to implement my own OTA stuff and possibly my own mqtt stuff and and and. I am not exactly a fan of the Arduino Framework, since it suggests all kinds of "look how easy this is" and it totally breaks down if you start to combine multiple libraries.

Oh well. Certainly not the problem of NeoPixelBus, so please feel free to close this issue  :)

closed 2022-01-02 by Makuna