Makuna/NeoPixelBus #560: flickering when ESP32 is using WiFi

beta-tester, 2022-02-07:
**Describe the bug**
i am using a ESP32 Dev board like this [link](https://joy-it.net/en/products/SBC-NodeMCU-ESP32).
when i use the NeoPixelBus library alone my LED strip performs well without visual glitche or flickering.
but when i also enable WiFi the LED strip begins to glitch and flicker

i also tried to put WiFi and the NeoPixelBus stuff in to different tasks on different CPUs but that does not prevent glitching or flickering.

**To Reproduce**
Steps to reproduce the behavior:
1. compile and upload the attached sketch to an ESP32
2. connect GPIO 13 of the ESP32 to  data-in the LED strip
3. see the flickering and glitches.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Development environment (please complete the following information):**
 - OS: Linux (Ubuntu PC) (RasPi OS Raspberry Pi), Windows (10 PC)
 - Build Environment : Arduino IDE 1.8.19, (Linux ARM32, Linux 64, Windows)
   Arduino-ESP32 2.0.2 via additional board manager URL
   https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
 - Board target [ESP32 Dev module](https://joy-it.net/en/products/SBC-NodeMCU-ESP32)
 - Library version 2.6.9 (installed via Arduino Library Manager).
 - LED strip with 50 RGB LEDs (in BRG configuration)

**Minimal Sketch that reproduced the problem:**
see attached my example `myLedStrio.ino` where you can see glicheing/flickering when WiFi is enabled.
i split the example in to two parts one `myLedStrip.cpp/.h` for the NeoPixelBus stuff and task, a simple rainbow animation.
and the second part `myOta.cpp/.h` for the WiFi stuff and task. as WiFi i used a WebService that expose two services http get and http post to realise OTA (over the air update).
- with comment out or rename `USE_OTA` definition, you can disable the use of WiFi.
- with comment out or rename `USE_MULTI_TASK` definition, you can disabling running things in separate tasks.

i tried different things:
- tried different modes for driving the LED strip
  NeoEsp32I2s1800KbpsMethod
  NeoEsp32Rmt0800KbpsMethod
  Neo800KbpsMethod
- i tried to put the NeoPixelBus part to CPU0 and the WiFi part to CPU1 and opposide

nothing helped to prevent glitches and flickering.

[myLedStrip.zip](https://github.com/Makuna/NeoPixelBus/files/8017945/myLedStrip.zip)

Makuna, 2022-02-07:
What LEDs are you using?  
Did you try specific methods for LEDs like `NeoEsp32I2s0Ws2812xMethod`?

beta-tester, 2022-02-08:
i am so sorry...
it looks like i have a power supply issue.
with another one with a bit higher output the issues are nearly gone.
in all configuration:
NeoEsp32I2s1800KbpsMethod
NeoEsp32Rmt0800KbpsMethod
Neo800KbpsMethod
NeoEsp32I2s0Ws2812xMethod

i will see if i can eliminate the last very few glitches, when i do 3.3V to 5.0V level shifting from the GPIO of ESP32 to Din of LED strip.

the non-USE_MULTI_TASK option i hacked in afterwards. i was using tasks in the initial "design" but hacked around to see how it looks as (kind of) single task. the waiting should not produce glitches when DMA is used i thought. it will just stop or slow down the animation.

**BTW: there is a dead link in your wiki:**
https://github.com/Makuna/NeoPixelBus/wiki/ESP32-NeoMethods
[Esp32 Dev Pin Map](https://github.com/espressif/arduino-esp32#esp32dev-board-pinmap)

maybe this is something similar:
https://docs.espressif.com/projects/arduino-esp32/en/latest/boards/ESP32-DevKitC-1.html#pin-layout