Makuna/NeoPixelBus #561: NeoEsp32Rmt0800KbpsMethod with ESP32-C3 wrong pixels mapping

mariusmotea, 2022-03-01:
**Describe the bug**
I use an animation that control one pixel at 100ms interval from pixel 0 to max pixels number. The bug is that the animation start from the declared number of pixels and continue from there. I my example i declared 60 x WS2812 leds and the animation start from pixel 61.

**To Reproduce**
Steps to reproduce the behavior:
Just flash my example sketch

**Expected behavior**
If the pixel 0 is declared i expect pixel 0 to be enabled.

**Development environment (please complete the following information):**
 - OS: Ubuntu 21.10
 - Build Environment Arduino IDE v.1.8.15
 - Board target ESP32C3 dev board 
 - Library version v2.6.9 

**Minimal Sketch that reproduced the problem:**
```
#include <WiFi.h>
#include <WiFiClient.h>
#include <WebServer.h>
#include <HTTPUpdateServer.h>
#include <NeoPixelBus.h>
#define ledGPIO 4
#define pixelCount 60
NeoPixelBus<NeoGrbFeature, NeoEsp32Rmt0800KbpsMethod>* strip = NULL;
RgbColor red = RgbColor(255, 0, 0);
RgbColor green = RgbColor(0, 255, 0);
RgbColor white = RgbColor(255);
RgbColor black = RgbColor(0);

void infoLight(RgbColor color) { // boot animation for leds count and wifi test
  // Flash the strip in the selected color. White = booted, green = WLAN connected, red = WLAN could not connect
  for (int i = 0; i < pixelCount; i++)
  {
    strip->SetPixelColor(i, color);
    strip->Show();
    delay(100);
    strip->SetPixelColor(i, black);
    strip->Show();
  }
}

void powerTheStrip() {
  pinMode(7, OUTPUT);
  delay(100);
  digitalWrite(7, HIGH);
}

WebServer httpServer(80);
HTTPUpdateServer httpUpdater;

void setup(void) {
  delay(1000);
  Serial.begin(115200);
  Serial.println();
  Serial.println("Booting Sketch...");
  pinMode(ledGPIO, OUTPUT);
  blinkLed(2, 200);
  pinMode(8, INPUT);
  pinMode(9, INPUT);
  WiFi.mode(WIFI_AP_STA);
  WiFi.begin(ssid, password);

  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    WiFi.begin(ssid, password);
    Serial.println("WiFi failed, retrying.");
  }

  httpUpdater.setup(&httpServer);
  httpServer.begin();
}

void loop(void) {
  httpServer.handleClient();
  if (digitalRead(8) == LOW) {
    blinkLed(3, 200);
    if (counter == 0) {
      powerTheStrip(); // Powering the strip using MOSFET
    }
    else if (counter == 1) {
      strip = new NeoPixelBus<NeoGrbFeature, NeoEsp32Rmt0800KbpsMethod>(60, 10); // and recreate with new count
      strip->Begin();
      infoLight(green);
    }
    else if (counter == 2) {
      infoLight(red);
    }
    else if (counter == 3) {
      infoLight(white);
    }
    else {
      for (int i = 0; i < pixelCount; i++)
      {
         Serial.println(String(i));
        strip->SetPixelColor(i, red);
        strip->Show();
        delay(100);
        strip->SetPixelColor(i, black);
        strip->Show();
      }
      ESP.restart();
    }
    counter += 1;
  } else if (digitalRead(9) == LOW) {
    blinkLed(4, 200);
  }
}
```
**Additional context**
Add any other context about the problem here.

if i use following code lines it display correctly, seams only in for lops the problem is manifesting. I also used Serial.println and the for loop show it stat from 0 and stop at 59 which is correct. 
```
      strip->SetPixelColor(0, green);
      strip->Show();
      delay(100);
      strip->SetPixelColor(0, black);
      strip->SetPixelColor(1, green);
      strip->Show();
      delay(100);
      strip->SetPixelColor(1, black);
      strip->SetPixelColor(2, green);
      strip->Show();
      delay(100);
      strip->SetPixelColor(2, black);
      strip->SetPixelColor(3, green);
      strip->Show();
      delay(100);
```

mariusmotea, 2022-03-01:
I investigate more and seams the issue is caused by second update of the same WS2812 led.

This is working file and first 4 leds of the strips are turned on with specific color:
```
  strip = new NeoPixelBus<NeoGrbFeature, NeoEsp32Rmt0800KbpsMethod>(60, 10); // and recreate with new count
  strip->Begin();
  strip->SetPixelColor(0, green);
  strip->Show();
  delay(100);
  strip->SetPixelColor(1, red);
  strip->Show();
  delay(100);
  strip->SetPixelColor(2, white);
  strip->Show();
  delay(100);
  strip->SetPixelColor(3, green);
  strip->Show();
  delay(1000);
```

this address just the first led, then continue from the led 61:
```
  strip = new NeoPixelBus<NeoGrbFeature, NeoEsp32Rmt0800KbpsMethod>(60, 10); // and recreate with new count
  strip->Begin();
  strip->SetPixelColor(0, green);
  strip->Show();
  delay(100);
  strip->SetPixelColor(0, black);
  strip->Show();
  strip->SetPixelColor(1, red);
  strip->Show();
  delay(100);
  strip->SetPixelColor(0, black);
  strip->Show();
  strip->SetPixelColor(2, white);
  strip->Show();
  delay(100);
  strip->SetPixelColor(0, black);
  strip->Show();
  strip->SetPixelColor(3, green);
  strip->Show();
  delay(1000);
```

Makuna, 2022-03-01:
This would have been a better as a question in the discussions area.

Your sketch contains all sorts of code that is not relevant to the issue.  If you truly believe it's a bug, you need to provide a minimal sketch that demonstrates the problem.  Often this enlightens the user to the core problem as you are discovering as you continue.

The loop happens very fast, you are reading if a pin is low to do some work, if this is a button, then there are better models of coding to handle this as what you are doing is problematic.  Search for button debounce.

There are several things that may not be obvious to you though with my library.  

When you call `Show()`, it will return before the data has been sent to the LEDs; as it is driven by hardware and is asynchronous.  This will be a problem when you call the `ESP.restart()` as it may not have finished sending the data before you restart.  To know its finished, you can call `CanShow()` which will return true if the data has been sent.

Calling `Show()` again is protected as my library will wait (looping on CanShow()) until the last one has been sent before updating the hardware to send another.  But the wire protocol requires a certain amount of "reset" time (no signal) to reset to the first LED in the strip and this "reset" time that `Show()` and `CanShow()` will wait is different per the LEDs and is defined as part of the Method.  Try using a specific LED model name for the Method, like `NeoEsp32Rmt0Ws2812xMethod`.  Make sure you truly have WS2812 LEDs, WS2812 is not the same as WS2812x where x could be almost any letter.  Check where you bought them.  They may say compatible, but not all have the exact specs, just close and this maybe an issue and is often resolved using a more specific Method.

Did you try the i2s method to compare if the issue is still present, like `NeoEsp32I2s0Ws2812xMethod`?

mariusmotea, 2022-03-02:
Hi,

I made the sketch more simple and and record a video to see exactly what happens:

https://www.youtube.com/watch?v=eMYJm_XWIac

The sketch is this:

```
#include <WiFi.h>
#include <WiFiClient.h>
#include <WebServer.h>
#include <HTTPUpdateServer.h>
#include <NeoPixelBus.h>

#define DATA_PIN     10
#define NUM_LEDS    60

const char* host = "esp32-webupdate";
const char* ssid = "MikroTik";
const char* password = "xxxxxxxxxxx";

RgbColor green = RgbColor(0, 255, 0);
RgbColor black = RgbColor(0);

NeoPixelBus<NeoGrbFeature, NeoEsp32Rmt0800KbpsMethod>* strip = NULL;

WebServer httpServer(80);
HTTPUpdateServer httpUpdater;

void setup(void) {
  delay(1000);
  pinMode(7, OUTPUT);
  digitalWrite(7, HIGH);
  WiFi.mode(WIFI_AP_STA);
  WiFi.begin(ssid, password);

  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    WiFi.begin(ssid, password);
    Serial.println("WiFi failed, retrying.");
  }

  strip = new NeoPixelBus<NeoGrbFeature, NeoEsp32Rmt0800KbpsMethod>(NUM_LEDS, DATA_PIN);
  strip->Begin();

  httpUpdater.setup(&httpServer);
  httpServer.begin();
}

void loop(void) {
    for (int i = 0; i < NUM_LEDS; i++)
  {
    httpServer.handleClient();
    strip->SetPixelColor(i, green);
    strip->Show();
    delay(50);
    strip->SetPixelColor(i, black);
    strip->Show();
  }
}
``` 

With 20 leds it is working fine, the issue strike when i increase the number of leds below 30.

Makuna, 2022-03-02:
Did you try `NeoEsp32Rmt0Ws2812xMethod`?

mariusmotea, 2022-03-03:
Thanks, `NeoEsp32Rmt0Ws2812xMethod` seams to be the fix for this code, i test it and the leds game play fine now. I believed ws2812 is default if not specified in the method.

closed 2022-03-03 by mariusmotea

Makuna, 2022-03-03:
I am slowly working on removing the 800Kbps and 400Kbps methods, the original 800Kbps method worked with WS2811b but they (LED manufacturers) changed the specs about four years ago and the newest ones require the WS2811x method instead.  That spec change was the start of model specific method names.