Makuna/NeoPixelBus #567: Support SM168xx Leds

Makuna, 2022-05-09:
Support Sunmoon Electronics SM168xx series of chips

**Describe the solution you'd like**
They require a tailing settings data to the stream
Mostly supported with now with the NeoSm168xxSettings and NeoRgbwSm168xxFeature, but the settings need to be ratified across the different chips and RGB versus RGBW.

**Additional context**
3 channel RGB
SM16803PB  1.8~60mA
SM16813P 1.75~19mA 
SM16823E 60~350mA
4 channel RGBW 
SM16804PB 1.8~19mA 
SM16804EB 1.5~60mA
SM16824E 60~350mA

Makuna, 2022-05-10:
Do we have a data sheet for the SM16813P?

Makuna, 2022-05-11:
https://github.com/Makuna/NeoPixelBus/pull/568
The newly exposed features and their corresponding settings classes are ...

- NeoRgbSm16803pbColorFeature with NeoSm16803pbSettings
- NeoRgbSm16823eColorFeature with NeoSm16823eSettings
- NeoRgbwSm16804ebColorFeature with NeoSm16804ebSettings
- NeoRgbwSm16824eColorFeature with NeoSm16824eSettings

The 03 and 04 just take the gain value for each channel, zero based (not 1 based like the specs)
NeoSm16804ebSettings(15,15,15,15);
The 23 and 24 take the gain values (zero based) and the resistor value in ohms.
NeoSm16824eSettings(15,15,15,15, 2200);

still requires testing.

Makuna, 2022-05-11:
Tested these four, latest merge fixed the gain order in the output and the calculation of the current from the gain.

Changchung, 2022-05-26:
I think that the 824 is RGBW. And it work just fine with wled.

Changchung, 2022-05-26:
> @Changchung the 824 is RGBW and the color feature name has it such 'NeoRgbwSm16824eColorFeature'.  What are you stating other than it works in WLED?

If you check your first comment show that the 823 is RGB and the third one it show that the 823 is RGBW. I confirm that the 824 is RGBW and 823 is RGB.

Changchung, 2022-05-26:
> @Changchung the 824 is RGBW and the color feature name has it such 'NeoRgbwSm16824eColorFeature'.  What are you stating other than it works in WLED?

No. Sorry. I miss understood your third comment.

Makuna, 2022-05-27:
v2.7.0

closed 2022-05-27 by Makuna