Makuna/NeoPixelBus #578: Compile fail: VARIANT_MCK not declared... using VSCode/PlatformIO, targeting board: RPi PICO

Barilium8, 2022-06-11:
**I have tried with just lib NeoPixelBus.h...**
#include <NeoPixelBus.h>

**And then tried...**
#include <SPI.h>
#include <NeoPixelBus.h>
#include <NeoPixelAnimator.h>

**Either way the VARIANT_MCK (and F_CPU) not declared...  errors occur.

Here is the beginning of the output form the compiler...**
> Executing task: C:\Users\steve\.platformio\penv\Scripts\platformio.exe run --target upload <
Processing pico (platform: raspberrypi; board: pico; framework: arduino)
---------------- Verbose mode can be enabled via `-v, --verbose` option ----------------
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pico.html
PLATFORM: Raspberry Pi RP2040 (1.7.0) > Raspberry Pi Pico
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, jlink, raspberrypi-swd)
PACKAGES:
 - framework-arduino-mbed @ 3.1.1
 - tool-openocd-raspberrypi @ 2.1100.0 (11.0)
 - tool-rp2040tools @ 1.0.2
 - toolchain-gccarmnoneeabi @ 1.90201.191206 (9.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 40 compatible libraries
Scanning dependencies...
Dependency Graph
|-- MIDI Library @ 5.0.2
|-- NeoPixelBus @ 2.7.0 
|   |-- SPI
|-- CLabs_Proto_Modules 
|-- SPI
Building in release mode
Compiling .pio\build\pico\src\main.cpp.o
Compiling .pio\build\pico\liba32\NeoPixelBus\internal\NeoGamma.cpp.o
Compiling .pio\build\pico\liba32\NeoPixelBus\internal\NeoPixelAnimator.cpp.o
In file included from .pio\libdeps\pico\NeoPixelBus\src/NeoPixelBus.h:123,
                 from .pio\libdeps\pico\NeoPixelBus\src\internal\NeoGamma.cpp:28:
.pio\libdeps\pico\NeoPixelBus\src/internal/NeoArmMethod.h:710:26: error: 'VARIANT_MCK' was not declared in this scope      
  710 | #define ARM_OTHER_SCALE  VARIANT_MCK / 2UL / 1000000UL
      |                          ^~~~~~~~~~~
.pio\libdeps\pico\NeoPixelBus\src/internal/NeoArmMethod.h:716:69: note: in expansion of macro 'ARM_OTHER_SCALE'
  716 |     static const uint32_t CyclesT0h = static_cast<uint32_t>((0.40 * ARM_OTHER_SCALE + 0.5) - (5 * ARM_OTHER_INST));
      |                                                                     ^~~~~~~~~~~~~~~
.pio\libdeps\pico\NeoPixelBus\src/internal/NeoArmMethod.h:711:33: error: 'F_CPU' was not declared in this scope
  711 | #define ARM_OTHER_INST   (2UL * F_CPU / VARIANT_MCK)
      |                                 ^~~~~
.pio\libdeps\pico\NeoPixelBus\src/internal/NeoArmMethod.h:716:99: note: in expansion of macro 'ARM_OTHER_INST'
  716 |     static const uint32_t CyclesT0h = static_cast<uint32_t>((0.40 * ARM_OTHER_SCALE + 0.5) - (5 * ARM_OTHER_INST));
      |                                                                                                   ^~~~~~~~~~~~~~   
.pio\libdeps\pico\NeoPixelBus\src/internal/NeoArmMethod.h:711:41: error: 'VARIANT_MCK' was not declared in this scope      
  711 | #define ARM_OTHER_INST   (2UL * F_CPU / VARIANT_MCK)
      |                                         ^~~~~~~~~~~
.pio\libdeps\pico\NeoPixelBus\src/internal/NeoArmMethod.h:716:99: note: in expansion of macro 'ARM_OTHER_INST'
  716 |     static const uint32_t CyclesT0h = static_cast<uint32_t>((0.40 * ARM_OTHER_SCALE + 0.5) - (5 * ARM_OTHER_INST));
      |                                                                                                   ^~~~~~~~~~~~~~   
.pio\libdeps\pico\NeoPixelBus\src/internal/NeoArmMethod.h:710:26: error: 'VARIANT_MCK' was not declared in this scope
  710 | #define ARM_OTHER_SCALE  VARIANT_MCK / 2UL / 1000000UL

Makuna, 2022-06-11:
This board is not supported by my library.   

The generic ARM implementation requires some basic support that is not present with this boards Arduino support.  This is common with MBED and SAMD boards support.

Further, this specific chip is dual core, meaning that more than likely the generic bit bang support even if ported would not work well and would require the use of a hardware feature specific to the chip/board.  

NOTES:
These are the ARDUINO_ARCH_ defined for this board when using the Arduin IDE with the Arduino (not SEEED) board support installed.
ARDUINO_ARCH_MBED
ARDUINO_ARCH_MBED_RP2040
ARDUINO_ARCH_RP2040

Makuna, 2022-06-29:
PIO - a feature of the RP2040 would allow for a bitbang code that is run within up to 8 internal IO CPUs (limited ASM instructions but good enough to do this) and will run these independent of the primary two cores.  This is the solution to use.

Makuna, 2024-03-11:
Should have something to show this next week.

Makuna, 2024-03-11:
https://github.com/Makuna/NeoPixelBus/pull/779

You can instance the NeoPixelBus four times on each PIO, for a theoretical eight channels.  But this will stop any other uses of PIO as there are only two PIO engines.  

Defaults to using PIO1 on IRQ 1.

Makuna, 2024-03-19:
https://github.com/Makuna/NeoPixelBus/releases/tag/2.7.9

closed 2024-03-19 by Makuna