Makuna/NeoPixelBus #579: Library does not compile when a ESP32-S3 board is selected in Arduino IDE

Miraculix200, 2022-06-11:
Library does not compile when a ESP32-S3 board is selected in Arduino IDE. This can be fixed by commenting out the line 115 in Neopixelbus.h:
`#include "internal/DotStarEsp32DmaSpiMethod.h"`

I guess some #if defined directives could be added. I think all ESP32-S3 boards include the compiler option -DARDUINO_VARIANT=\"esp32s3\"

Makuna, 2022-06-11:
If they follow their standards, it should be `CONFIG_IDF_TARGET_ESP32S3`

Miraculix200, 2022-06-12:
Datasheet says the ESP32-S3 supports 4 RX and 4 TX channels for RMT. DMA TX access on channel 3, RX access on channel 7. This RMT Neopixel example supposedly works with ESP32-S3 (I did not test it): https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/RMT/RMTWriteNeoPixel/RMTWriteNeoPixel.ino

Makuna, 2022-06-12:
https://github.com/Makuna/NeoPixelBus/pull/580

Try it now.

Miraculix200, 2022-06-13:
It compiles without errors and works. I'm using the NeoEsp32Rmt0800KbpsMethod though, not SPI.

Makuna, 2022-06-13:
I2s is tracked by https://github.com/Makuna/NeoPixelBus/issues/559, its currently not supported for C3 nor S3.

Makuna, 2022-06-21:
I am leaving this open until I create a released version that includes the changes.  I marked it Pending, this means the github has the changes, but they haven't been officially released.  Sort of wish I had a way for the author (you) to mark it as "I agree its fixed" without closing it.

Makuna, 2023-02-03:
https://github.com/Makuna/NeoPixelBus/releases/tag/2.7.2

closed 2023-02-03 by Makuna