Makuna/NeoPixelBus #584: Arduino Nano BLE conflict with multiple hardware pwm?

silvan2468, 2022-07-20:
Hy there

I try to use 2x4 hardware pwms in combination with the NeoPixelBus which uses also hardware PWM DMA.
Generating the pwm signals work fine. Also control the Neopixels with only 1x4 hardware pwm signals works fine (as long I only use pwm-block 0.
The problem is, as soon I add a second pwm block (doesn't matter if block 1, 2 or 3), the Neopixels flicker uncontrolled twice in a minute.
It also doesn't matter if I change NeoPixelBus hardware pwm port by 
"NeoPixelBus<NeoGrbwFeature, NeoNrf52xPwm1800KbpsMethod> strip(PixelCount, PixelPin);"

Here's my code:

```
#include "nrfx_pwm.h"
#include <NeoPixelBus.h>

const uint16_t PixelCount = 10; // this example assumes 4 pixels, making it smaller will cause a failure
const uint8_t PixelPin = 5;  // make sure to set this to the correct pin, ignored for Esp8266
#define colorSaturation 10
NeoPixelBus<NeoGrbwFeature, Neo800KbpsMethod> strip(PixelCount, PixelPin);
//NeoPixelBus<NeoGrbwFeature, NeoNrf52xPwm1800KbpsMethod> strip(PixelCount, PixelPin);
RgbColor red(colorSaturation, 0, 0);
RgbColor green(0, colorSaturation, 0);
RgbColor blue(0, 0, colorSaturation);
RgbColor white(colorSaturation);
RgbColor black(0);

static nrfx_pwm_t pwm0 = NRFX_PWM_INSTANCE(0);
static nrfx_pwm_t pwm1 = NRFX_PWM_INSTANCE(1);

static nrf_pwm_values_individual_t seq0_values[] = {0, 0, 0, 0};
static nrf_pwm_values_individual_t seq1_values[] = {0, 0, 0, 0};

static nrf_pwm_sequence_t seq0 = {
    .values = {
        .p_individual = seq0_values
    },
    .length          = NRF_PWM_VALUES_LENGTH(seq0_values),
    .repeats         = 1,
    .end_delay       = 0
};

static nrf_pwm_sequence_t seq1 = {
    .values = {
        .p_individual = seq1_values
    },
    .length          = NRF_PWM_VALUES_LENGTH(seq1_values),
    .repeats         = 1,
    .end_delay       = 0
};

void setup() {

  strip.Begin();
  strip.Show();

  nrfx_pwm_config_t config0 = {
        .output_pins  = {
            32+ 11, // Arduino pin 2  - 1.11
            27,     // Arduino pin 9  - 0.27
            30,     // Arduino pin 16 - 0.30
            29,     // Arduino pin 17 - 0.29
        },
        .irq_priority = 7,
        .base_clock   = NRF_PWM_CLK_8MHz,
        .count_mode   = NRF_PWM_MODE_UP,
        .top_value    = 256,                        // --> 31'250 Hz; Auflösung = 256 Steps
        .load_mode    = NRF_PWM_LOAD_INDIVIDUAL,
        .step_mode    = NRF_PWM_STEP_AUTO,
    };

  nrfx_pwm_init(&pwm0, &config0, NULL);
  nrfx_pwm_config_t config1 = {
        .output_pins  = {
            32+ 12, // Arduino pin 3  - 1.12
            32+ 13, // Arduino pin 4  - 1.15
            21,     // Arduino pin 8  - 0.21
            32+ 2,  // Arduino pin 10 - 1.02           
        },
        .irq_priority = 7,
        .base_clock   = NRF_PWM_CLK_1MHz,
        .count_mode   = NRF_PWM_MODE_UP,
        .top_value    = 4000,                     // --> 250 Hz = 4ms; Auflösung = 1 ys
        .load_mode    = NRF_PWM_LOAD_INDIVIDUAL,
        .step_mode    = NRF_PWM_STEP_AUTO,
    };

  nrfx_pwm_init(&pwm1, &config1, NULL);

  (*seq0_values).channel_0 = 50   | 0x8000;     // 0x8000 inverts the PWM signal
  (*seq0_values).channel_1 = 100  | 0x8000;
  (*seq0_values).channel_2 = 150  | 0x8000;
  (*seq0_values).channel_3 = 200  | 0x8000;
  (void)nrfx_pwm_simple_playback(&pwm0, &seq0, 1, NRFX_PWM_FLAG_LOOP);

  (*seq1_values).channel_0 = 2000  | 0x8000;     // 0x8000 inverts the PWM signal
  (*seq1_values).channel_1 = 100   | 0x8000;
  (*seq1_values).channel_2 = 500   | 0x8000;
  (*seq1_values).channel_3 = 1000  | 0x8000;
//  (void)nrfx_pwm_simple_playback(&pwm1, &seq1, 1, NRFX_PWM_FLAG_LOOP);
}

void loop() {
  strip.SetPixelColor(0, red);
  strip.SetPixelColor(1, green);
  strip.SetPixelColor(2, blue);
  strip.SetPixelColor(3, RgbwColor(colorSaturation));
  strip.Show();
  delay(1000);
  // turn off the pixels
  strip.SetPixelColor(0, black);
  strip.SetPixelColor(1, black);
  strip.SetPixelColor(2, black);
  strip.SetPixelColor(3, black);
  strip.Show();
  delay(1000);
}
```

As soon I uncoment this line:
 (void)nrfx_pwm_simple_playback(&pwm1, &seq1, 1, NRFX_PWM_FLAG_LOOP);
the LEDs start doing false :(

Any idea what the problem may be?
Because the pwm signals looks fine, I think the problem may be at the NeoPixelBus library.

Thanks a lot and kind regards
Silvan

Makuna, 2022-07-22:
The default PWM unit for NeoPixelBus is 2 (0-3) unless you specify otherwise.

I use the predefined NRF_PWM0, NRF_PWM1, NRF_PWM2, NRF_PWM3, which are `NRF_PWM_Type` and directly modify their members.  There is nothing outside each context is changed, so as long as you different units then they should be isolated.  

I have no clue about the API NRFX_PWM_INSTANCE() or the nrfx_pwm_* calls as it is not what I use.  You may want to look into it more to see what it does.

Makuna, 2022-07-22:
Try the `NeoNrf52xPwm3Ws2812xMethod` method for the neopixels.
Also try moving the strip.Begin() to the end of Setup().

A quick glance at the APIs you are using for PWM confirms to me that I should not use them.  It's just a layer, that I am unsure what they are trying to solve with it, as it makes it more obtuse to use and rather overcomplicated.

silvan2468, 2022-07-22:
Hy Makuna
Thanks a lot for your help. Unfortunately, both did not help. The idea to put the begin() at the end of the setup I also already had.
I don't know what this NRFX API does. I just copied from a forum and for generating the PWM signals this looks fine.

I tested right now the library "nrf mbed slow pwm" from khoih. Probably I will go this way. Also because with the Nano BLE it is told that only timer 3 and 4 are "free". This is disastrous. So I need to save timers wherever possible.

But I will also have a look to NRF_PWMx. Perhaps I can do generate the hardware pwm this way. Or do you know perhaps already a PWM library for the Nano BLE which does so?

Thanks a lot again and have a nice weekend
Silvan

silvan2468, 2022-07-23:
Just one more question I funnywise didn't found by googleing: does hardware pwm needs one of the hardware timers 0-4? The only thing I found in the Nrf52840 online datasheet was, that the 4 pwm groups share one common 15 bit counter. But timers 0-5 are 32bit timers if I have seen right. So usage uf hardware pwms doesn't use hardware timers but has it own counter? 
This would be genial, but does not match with my experience with every other microcontrollers I have.

silvan2468, 2022-07-23:
I just found, that every "analogWrite" from arduino uses one of the 4 hardware pwm groups. 
https://forum.arduino.cc/t/arduino-nano-33-ble-high-frequency-pwm-for-more-than-4-pins/648970/4

So if using hardware pwm would affect the hardware timers, every usage of analogWrite would use/occupie a jardware timer what not can be. So I am quite sure hardware pwm does not conflict/use hardware timers. 

So I think our/my problem is not caused by hardware timers. It has to be something conflicting at the pwm groups we use.

silvan2468, 2022-07-23:
Oh dear!!!
I found the problem!
I tested so many things which didn't work for so many hours!
At least I did an update to the newest mbed version (3.2.0; unfortunately I don't know which version I had before) and now it works perfect!!!! Together with 3x4 = 12 hardware pwms!
So problem was in mbed. Once again :(
Why do I buy Arduinos? Ah yes because it is soooo easy, well tested and safe to use! ;)

Here is my complete code because I think this can be very helpful for some other people:

```
#include "nrfx_pwm.h"
#include <NeoPixelBus.h>

const uint16_t PixelCount = 10; // this example assumes 4 pixels, making it smaller will cause a failure
const uint8_t PixelPin = 5; // make sure to set this to the correct pin, ignored for Esp8266
#define colorSaturation 10

NeoPixelBus<NeoGrbwFeature, NeoNrf52xPwm3Sk6812Method> strip(PixelCount, PixelPin);
//NeoPixelBus<NeoGrbwFeature, Neo800KbpsMethod> strip(PixelCount, PixelPin);
//NeoPixelBus<NeoGrbwFeature, NeoNrf52xPwm1800KbpsMethod> strip(PixelCount, PixelPin);
RgbColor red(colorSaturation, 0, 0);
RgbColor green(0, colorSaturation, 0);
RgbColor blue(0, 0, colorSaturation);
RgbColor white(colorSaturation);
RgbColor black(0);

static nrfx_pwm_t pwm0 = NRFX_PWM_INSTANCE(0);
static nrfx_pwm_t pwm1 = NRFX_PWM_INSTANCE(1);
static nrfx_pwm_t pwm2 = NRFX_PWM_INSTANCE(2);

static nrf_pwm_values_individual_t seq0_values[] = {0, 0, 0, 0};
static nrf_pwm_values_individual_t seq1_values[] = {0, 0, 0, 0};
static nrf_pwm_values_individual_t seq2_values[] = {0, 0, 0, 0};

static nrf_pwm_sequence_t seq0 = {
.values = {
.p_individual = seq0_values
},
.length = NRF_PWM_VALUES_LENGTH(seq0_values),
.repeats = 0,
.end_delay = 0
};

static nrf_pwm_sequence_t seq1 = {
.values = {
.p_individual = seq1_values
},
.length = NRF_PWM_VALUES_LENGTH(seq1_values),
.repeats = 0,
.end_delay = 0
};

static nrf_pwm_sequence_t seq2 = {
.values = {
.p_individual = seq2_values
},
.length = NRF_PWM_VALUES_LENGTH(seq2_values),
.repeats = 0,
.end_delay = 0
};

uint16_t i = 1000;

void setup() {

nrfx_pwm_config_t config0 = {
.output_pins = {
32+ 11, // Arduino pin 2 - 1.11
27, // Arduino pin 9 - 0.27
30, // Arduino pin 16 - 0.30
29, // Arduino pin 17 - 0.29
},
.irq_priority = 7,
.base_clock = NRF_PWM_CLK_8MHz,
.count_mode = NRF_PWM_MODE_UP,
.top_value = 256, // --> 31'250 Hz; Auflösung = 256 Steps
.load_mode = NRF_PWM_LOAD_INDIVIDUAL,
.step_mode = NRF_PWM_STEP_AUTO,
};
nrfx_pwm_init(&pwm0, &config0, NULL);

nrfx_pwm_config_t config1 = {
.output_pins = {
32+ 12, // Arduino pin 3 - 1.12
32+ 15, // Arduino pin 4 - 1.15
21, // Arduino pin 8 - 0.21
32+ 2, // Arduino pin 10 - 1.02
// NRFX_PWM_PIN_NOT_USED,
},
.irq_priority = 7,
.base_clock = NRF_PWM_CLK_1MHz,
.count_mode = NRF_PWM_MODE_UP,
.top_value = 4000, // --> 250 Hz = 4ms; Auflösung = 1 ys
.load_mode = NRF_PWM_LOAD_INDIVIDUAL,
.step_mode = NRF_PWM_STEP_AUTO,
};
nrfx_pwm_init(&pwm1, &config1, NULL);

nrfx_pwm_config_t config2 = {
.output_pins = {
32+ 10, // Arduino pin 0 - 1.10
32+ 3, // Arduino pin 1 - 1.03
23, // Arduino pin 7 - 0.23
32+ 14, // Arduino pin 6 - 1.14
},
.irq_priority = 7,
.base_clock = NRF_PWM_CLK_1MHz,
.count_mode = NRF_PWM_MODE_UP,
.top_value = 20000, // --> 50 Hz = 20ms; resolution = 1 ys
.load_mode = NRF_PWM_LOAD_INDIVIDUAL,
.step_mode = NRF_PWM_STEP_AUTO,
};
nrfx_pwm_init(&pwm2, &config2, NULL);

(*seq0_values).channel_0 = 50 | 0x8000; // 0x8000 inverts the PWM signal
(*seq0_values).channel_1 = 100 | 0x8000;
(*seq0_values).channel_2 = 150 | 0x8000;
(*seq0_values).channel_3 = 200 | 0x8000;
(void)nrfx_pwm_simple_playback(&pwm0, &seq0, 1, NRFX_PWM_FLAG_LOOP);

(*seq1_values).channel_0 = i | 0x8000; // 0x8000 inverts the PWM signal
(*seq1_values).channel_1 = 100 | 0x8000;
(*seq1_values).channel_2 = 500 | 0x8000;
(*seq1_values).channel_3 = 1000 | 0x8000;
(void)nrfx_pwm_simple_playback(&pwm1, &seq1, 1, NRFX_PWM_FLAG_LOOP);

(*seq2_values).channel_0 = i | 0x8000; // 0x8000 inverts the PWM signal
(*seq2_values).channel_1 = 100 | 0x8000;
(*seq2_values).channel_2 = 500 | 0x8000;
(*seq2_values).channel_3 = 1000 | 0x8000;
(void)nrfx_pwm_simple_playback(&pwm2, &seq2, 1, NRFX_PWM_FLAG_LOOP);

strip.Begin();
strip.Show();
}

void loop() {
// turn on the pixels
strip.SetPixelColor(0, red);
strip.SetPixelColor(1, green);
strip.SetPixelColor(2, blue);
strip.SetPixelColor(3, RgbwColor(colorSaturation));
strip.Show();
delay(1000);
// turn off the pixels
strip.SetPixelColor(0, black);
strip.SetPixelColor(1, black);
strip.SetPixelColor(2, black);
strip.SetPixelColor(3, black);
strip.Show();
delay(1000);

// update servo pwm
(*seq2_values).channel_0 = i | 0x8000;
i += 100;
if(i >2000){i = 1000;}
(void)nrfx_pwm_simple_playback(&pwm2, &seq2, 1, NRFX_PWM_FLAG_LOOP);
}
```

closed 2022-07-23 by silvan2468