Makuna/NeoPixelBus #586: v2.7 (and probably earlier) not compatible with older ESP cores (2.4.2 / 2.7.4)

devarishi7, 2022-08-03:
When Compiling with v2.7 for Generic ESP8266 using core 2.4.2 I get a whole heap of error messages. When using ESP core 3.0.2 these are not there and everything compiles just fine.
While using an older version of Neopixelbus (don't know which one actually couldn't retrieve that info somehow) these do not show up.
`Arduino: 1.8.19 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, Flash, ck, 26 MHz, 40MHz, QIO, 512K (no SPIFFS), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

In file included from C:\Arduino Files\Sketches\libraries\NeoPixelBus-master\src/internal/NeoEsp8266DmaMethod.h:34:0,

                 from C:\Arduino Files\Sketches\libraries\NeoPixelBus-master\src/NeoPixelBus.h:105,

                 from C:\Arduino Files\Sketches\ESP_Sketches\DumbSeq\DumbSeq_v113\DumbSeq_v113.ino:7:

C:\Arduino Files\Sketches\libraries\NeoPixelBus-master\src/internal/NeoEsp8266I2sMethodCore.h:105:17: error: variable or field 'IRAM_ATTR' declared void

     static void IRAM_ATTR i2s_slc_isr(void)

                 ^

C:\Arduino Files\Sketches\libraries\NeoPixelBus-master\src/internal/NeoEsp8266I2sMethodCore.h:105:17: error: expected ';' at end of member declaration

C:\Arduino Files\Sketches\libraries\NeoPixelBus-master\src/internal/NeoEsp8266I2sMethodCore.h:105:43: error: ISO C++ forbids declaration of 'i2s_slc_isr' with no type [-fpermissive]

     static void IRAM_ATTR i2s_slc_isr(void)

                                           ^

In file included from C:\Users\Rishi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/pgmspace.h:10:0,

                 from C:\Users\Rishi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/Arduino.h:256,

                 from sketch\DumbSeq_v113.ino.cpp:1:

C:\Arduino Files\Sketches\libraries\NeoPixelBus-master\src/internal/NeoEsp8266I2sMethodCore.h: In member function 'void NeoEsp8266I2sMethodCore::InitializeI2s(uint32_t, uint32_t)':

C:\Users\Rishi\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/include/ets_sys.h:170:54: error: invalid use of member function (did you forget the '()' ?)

     ets_isr_attach(ETS_SLC_INUM, (int_handler_t)(func), (void *)(arg))

                                                      ^

C:\Arduino Files\Sketches\libraries\NeoPixelBus-master\src/internal/NeoEsp8266I2sMethodCore.h:269:9: note: in expansion of macro 'ETS_SLC_INTR_ATTACH'

         ETS_SLC_INTR_ATTACH(i2s_slc_isr, NULL);

         ^

In file included from C:\Arduino Files\Sketches\libraries\NeoPixelBus-master\src/NeoPixelBus.h:105:0,

                 from C:\Arduino Files\Sketches\ESP_Sketches\DumbSeq\DumbSeq_v113\DumbSeq_v113.ino:7:

C:\Arduino Files\Sketches\libraries\NeoPixelBus-master\src/internal/NeoEsp8266DmaMethod.h: At global scope:

C:\Arduino Files\Sketches\libraries\NeoPixelBus-master\src/internal/NeoEsp8266DmaMethod.h:214:29: error: expected unqualified-id before '[' token

     NeoEsp8266DmaMethodBase([[maybe_unused]] uint8_t pin, uint16_t pixelCount, size_t elementSize, size_t settingsSize) : 

                             ^

C:\Arduino Files\Sketches\libraries\NeoPixelBus-master\src/internal/NeoEsp8266DmaMethod.h:214:29: error: expected ')' before '[' token

C:\Arduino Files\Sketches\libraries\NeoPixelBus-master\src/internal/NeoEsp8266DmaMethod.h:251:10: error: variable or field 'IRAM_ATTR' declared void

     void IRAM_ATTR Update(bool)

          ^

C:\Arduino Files\Sketches\libraries\NeoPixelBus-master\src/internal/NeoEsp8266DmaMethod.h:251:10: error: expected ';' at end of member declaration

C:\Arduino Files\Sketches\libraries\NeoPixelBus-master\src/internal/NeoEsp8266DmaMethod.h:251:31: error: ISO C++ forbids declaration of 'Update' with no type [-fpermissive]

     void IRAM_ATTR Update(bool)

[... 129 lines cut ...]

C:\Arduino Files\Sketches\libraries\NeoPixelBus-master\src/internal/NeoEsp8266I2sDmx512Method.h:123:5: note: NeoEsp8266I2sDmx512MethodBase<T_SPEED>::NeoEsp8266I2sDmx512MethodBase(uint16_t, size_t, size_t) [with T_SPEED = NeoEsp8266I2sDmx512Speed; uint16_t = short unsigned int; size_t = unsigned int]

     NeoEsp8266I2sDmx512MethodBase(uint16_t pixelCount, size_t elementSize, size_t settingsSize) :

     ^

C:\Arduino Files\Sketches\libraries\NeoPixelBus-master\src/internal/NeoEsp8266I2sDmx512Method.h:123:5: note:   candidate expects 3 arguments, 4 provided

C:\Arduino Files\Sketches\libraries\NeoPixelBus-master\src/internal/NeoEsp8266I2sDmx512Method.h:118:34: note: constexpr NeoEsp8266I2sDmx512MethodBase<NeoEsp8266I2sDmx512Speed>::NeoEsp8266I2sDmx512MethodBase(const NeoEsp8266I2sDmx512MethodBase<NeoEsp8266I2sDmx512Speed>&)

 template<typename T_SPEED> class NeoEsp8266I2sDmx512MethodBase : NeoEsp8266I2sMethodCore

                                  ^

C:\Arduino Files\Sketches\libraries\NeoPixelBus-master\src/internal/NeoEsp8266I2sDmx512Method.h:118:34: note:   candidate expects 1 argument, 4 provided

exit status 1

Error compiling for board Generic ESP8266 Module.

`
Some appear to be related to IRAM_ATTR which is undefined in the older cores, but there are also some other which don't make much sense to me.
Anyway as said using an older Neopixelbus version with the older cores works.
Using Neopixelbus 2.7 only works with core 3.0.2 (maybe a few more but that is what i tested)

Makuna, 2022-08-03:
https://github.com/Makuna/NeoPixelBus/commit/0362fcda3143d2095059d0bd3f98a27e343ddc1a

- ICACHE_RAM_ATTR was changed to the standard IRAM_ATTR in the core.
- attach interrupt I believe they added another argument.
- Older compiler doesn't support the `[[maybe_unused]]` attribute (c17++ or newer if I remember correctly)

I just don't have the personal bandwith to be backward compatible when things change in cores, so to use older cores requires that you use older library versions or branch and make the changes to a local version yourself.

devarishi7, 2022-08-04:
Yes understand. Well i am going to see if i can mange to get the latest version to compile with an older core, while maintaining compatibility.
For now i have added
`#ifndef IRAM_ATTR
  #define IRAM_ATTR ICACHE_RAM_ATTR
#endif
`
to first included file for ARDUINO_ARCH_ESP8266 (somehow nested Pre-compiler directives is not my specialty)
which gets rid of that part.
And comparing to older versions i will see if i can get rid of the rest of the errors.

Makuna, 2022-08-04:
My suggestion is to add a compatibility file (ex: Esp8266_Core_Comp.h) that you include all these things into, then inside add the statements like you mentioned inside.  Then in every esp8266 specific method .h file in neopixelbus, include this header file.
Other hints:  
- see line 53 in NeoEsp8266I2sMethodCore.h for example of checking core versions.

devarishi7, 2022-08-05:
Seriously ? Look i am just trying to get it to work, Adding the IRAM_ATTR was easy enough, removing the [[maybe unused]] also didn't cause any issue, but within the bitbangmethod there are IRAM_ATTR in a cpp file which are being ignored and were moved to a 'C' file to prevent that, well that is a bit beyond me though. When i just remove the reference to the .h file and remove the .cpp file, but now a cryptic message shows up
`sketch\DumbSeq_v113.ino.cpp.o: In function `StringSumHelper':
C:\Arduino Files\Sketches\ESP_Sketches\DumbSeq\DumbSeq_v113/WebPages.ino:2091: undefined reference to `NeoUtil::Reverse8BitsLookup'
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board Generic ESP8285 Module.
`
I decided to first 'uncomment' the include to the header file again, Move the functions to a .c file, comment out the contents 
 of the functions (i don't need bitbang method) , but that didn't help and the udefined reference is still showing up. So time to have some rest. These things are hard enough within my own code, but in such an extensive project, it must be near impossible. And there is so much stuff i never use, it just doesn't make much sense to me anymore. I will keep recommending this library, but with the note that it will only work with the recent cores, which i guess for most people is fine.

Makuna, 2022-08-05:
keep documenting what you find, it will help reduce the effort for me when I can find the time to try to solve this.

devarishi7, 2022-08-05:
Yeah no worries, well i got it sorted, and it works for core 2.4.2, but basically i think it is the same for any core before 3.0.0
- The definition of IRAM_ATTR that is missing
- The removing of the [[maybe unused]] tag, there's probably an easier / faster way then just removing them
- The ICACHE_IRAM_ATTR that is being ignored in .cpp files in the Esp8266bitbangMethod.cpp and should be moved to a .c file
- And the Lookup table in NeoUtil.h which i changed to the following

`
private:
    /*static constexpr uint8_t Reverse8BitsLookup[16] = {
            0x0, 0x8, 0x4, 0xc, 0x2, 0xa, 0x6, 0xe,
            0x1, 0x9, 0x5, 0xd, 0x3, 0xb, 0x7, 0xf };*/
    

public:
    inline static uint8_t Reverse8Bits(uint8_t n)
    {
		const uint8_t Reverse8BitsLookup[16] = {
            0x0, 0x8, 0x4, 0xc, 0x2, 0xa, 0x6, 0xe,
            0x1, 0x9, 0x5, 0xd, 0x3, 0xb, 0x7, 0xf };
        return (Reverse8BitsLookup[n & 0b1111] << 4) | Reverse8BitsLookup[n >> 4];
    }

`

h0jeZvgoxFepBQ2C, 2024-01-25:
I have the same problems:

```
In file included from /Users/myuser/Documents/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/methods/NeoEsp8266DmaMethod.h:34:0,
                 from /Users/myuser/Documents/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoMethods.h:48,
                 from /Users/myuser/Documents/Arduino/libraries/NeoPixelBus_by_Makuna/src/NeoPixelBus.h:43,
                 from /Users/myuser/Documents/Arduino/sketch_usb_rgb_controller/sketch_usb_rgb_controller.ino:14:
/Users/myuser/Documents/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/methods/NeoEsp8266I2sMethodCore.h:116:17: error: variable or field 'IRAM_ATTR' declared void
     static void IRAM_ATTR i2s_slc_isr(void)
                 ^
/Users/myuser/Documents/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/methods/NeoEsp8266I2sMethodCore.h:116:17: error: expected ';' at end of member declaration
/Users/myuser/Documents/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/methods/NeoEsp8266I2sMethodCore.h:116:43: error: ISO C++ forbids declaration of 'i2s_slc_isr' with no type [-fpermissive]
     static void IRAM_ATTR i2s_slc_isr(void)
                                           ^
In file included from /Users/myuser/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0-beta2/cores/esp8266/pgmspace.h:11:0,
                 from /Users/myuser/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0-beta2/cores/esp8266/Arduino.h:258,
                 from /private/var/folders/sf/c2pjz6v52tbc6zkw1fsrx9kc0000gn/T/arduino/sketches/0D096AFAC6607C1E1AE0F59B5459E124/sketch/sketch_usb_rgb_controller.ino.cpp:1:
/Users/myuser/Documents/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/methods/NeoEsp8266I2sMethodCore.h: In member function 'void NeoEsp8266I2sMethodCore::InitializeI2s(uint32_t, uint32_t)':
/Users/myuser/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0-beta2/tools/sdk/include/ets_sys.h:170:54: error: invalid use of member function (did you forget the '()' ?)
     ets_isr_attach(ETS_SLC_INUM, (int_handler_t)(func), (void *)(arg))
                                                      ^
/Users/myuser/Documents/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/methods/NeoEsp8266I2sMethodCore.h:294:9: note: in expansion of macro 'ETS_SLC_INTR_ATTACH'
         ETS_SLC_INTR_ATTACH(i2s_slc_isr, NULL);
         ^
In file included from /Users/myuser/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0-beta2/cores/esp8266/Arduino.h:38:0,
                 from /private/var/folders/sf/c2pjz6v52tbc6zkw1fsrx9kc0000gn/T/arduino/sketches/0D096AFAC6607C1E1AE0F59B5459E124/sketch/sketch_usb_rgb_controller.ino.cpp:1:
/Users/myuser/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0-beta2/cores/esp8266/esp8266_peri.h:32:134: error: 'rom_i2c_writeReg_Mask' was not declared in this scope
 #define i2c_writeReg_Mask(block, host_id, reg_add, Msb, Lsb, indata)  rom_i2c_writeReg_Mask(block, host_id, reg_add, Msb, Lsb, indata)
                                                                                                                                      ^
/Users/myuser/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0-beta2/cores/esp8266/esp8266_peri.h:33:55: note: in expansion of macro 'i2c_writeReg_Mask'
 #define i2c_writeReg_Mask_def(block, reg_add, indata) i2c_writeReg_Mask(block, block##_hostid,  reg_add,  reg_add##_msb,  reg_add##_lsb,  indata)
                                                       ^
/Users/myuser/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0-beta2/cores/esp8266/esp8266_peri.h:756:43: note: in expansion of macro 'i2c_writeReg_Mask_def'
 #define I2S_CLK_ENABLE()                  i2c_writeReg_Mask_def(i2c_bbpll, i2c_bbpll_en_audio_clock_out, 1)
                                           ^
/Users/myuser/Documents/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/methods/NeoEsp8266I2sMethodCore.h:303:9: note: in expansion of macro 'I2S_CLK_ENABLE'
         I2S_CLK_ENABLE();
         ^
In file included from /Users/myuser/Documents/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoMethods.h:48:0,
                 from /Users/myuser/Documents/Arduino/libraries/NeoPixelBus_by_Makuna/src/NeoPixelBus.h:43,
                 from /Users/myuser/Documents/Arduino/sketch_usb_rgb_controller/sketch_usb_rgb_controller.ino:14:
/Users/myuser/Documents/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/methods/NeoEsp8266DmaMethod.h: At global scope:
/Users/myuser/Documents/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/methods/NeoEsp8266DmaMethod.h:293:29: error: expected unqualified-id before '[' token
     NeoEsp8266DmaMethodBase([[maybe_unused]] uint8_t pin, uint16_t pixelCount, size_t elementSize, size_t settingsSize) : 
                             ^
/Users/myuser/Documents/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/methods/NeoEsp8266DmaMethod.h:293:29: error: expected ')' before '[' token
/Users/myuser/Documents/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/methods/NeoEsp8266DmaMethod.h:330:10: error: variable or field 'IRAM_ATTR' declared void
     void IRAM_ATTR Update(bool)
          ^
/Users/myuser/Documents/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/methods/NeoEsp8266DmaMethod.h:330:10: error: expected ';' at end of member declaration
[... cut ...]
/Users/myuser/Documents/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/methods/NeoEsp8266DmaMethod.h:265:36: note: constexpr NeoEsp8266DmaMethodBase<NeoEsp8266DmaEncode<NeoEsp8266DmaSpeedWs2812x> >::NeoEsp8266DmaMethodBase(const NeoEsp8266DmaMethodBase<NeoEsp8266DmaEncode<NeoEsp8266DmaSpeedWs2812x> >&)
 template<typename T_ENCODER> class NeoEsp8266DmaMethodBase : NeoEsp8266I2sMethodCore
                                    ^
/Users/myuser/Documents/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/methods/NeoEsp8266DmaMethod.h:265:36: note:   candidate expects 1 argument, 4 provided

exit status 1

Compilation error: exit status 1
```

Generic ESP8266 module

h0jeZvgoxFepBQ2C, 2024-01-25:
Edit: Works fine after updating the esp8266 arduino library!