Makuna/NeoPixelBus #587: Build error for Mega2560

furynick, 2022-08-06:
Building NeoPixelTopologyTest failed for Mega2560 with fresh install of IDE and latest code from github.

```
Arduino : 1.8.19 (Windows 7), Carte : "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"
In file included from C:\Users\nico\Documents\Arduino\libraries\NeoPixelBus\src/NeoPixelAnimator.h:30:0,

                 from C:\Users\nico\Documents\Arduino\libraries\NeoPixelBus\examples\topologies\NeoPixelTopologyTest\NeoPixelTopologyTest.ino:10:

C:\Users\nico\Documents\Arduino\libraries\NeoPixelBus\src/internal/NeoEase.h:37:10: fatal error: functional: No such file or directory

 #include <functional>

          ^~~~~~~~~~~~

compilation terminated.

exit status 1
```

I didn't see any issue on this error and no prerequisites info on wiki & readme. Am I missing something ?

Makuna, 2022-08-07:
Not sure what changed, but the change is simple.  In the ino, swap the order of includes too...
```
#include <NeoPixelBus.h>
#include <NeoPixelAnimator.h>
```

furynick, 2022-08-07:
Build successful with swapped includes.

Almost all examples under topologies folder have the same issue.

Thanks for your help.

Makuna, 2022-08-12:
https://github.com/Makuna/NeoPixelBus/pull/593

Makuna, 2023-02-03:
https://github.com/Makuna/NeoPixelBus/releases/tag/2.7.2

closed 2023-02-03 by Makuna