Makuna/NeoPixelBus #613: Support for LGT8F328P

roytomeij, 2023-01-13:
The LGT8F328P-LQFP32 and LGT8F328P-LQFP48 boards are broadly advertised as "Arduino Nano replacement boards." They seem like nice boards with more PWM-capable pins and a faster processor. The latter is an issue, though: when trying to compile an Arduino sketch using NeoPixelBus, it throws this error:

```NeoPixelBus/src/internal/NeoAvrMethod.h:68:2: error: #error "CPU SPEED NOT SUPPORTED"```

Without knowing anything about building Arduino libraries: it would be very helpful to add support for this board. I'd be happy to test.

Makuna, 2023-01-13:
What frequency are you selecting in the boards menu?  It seems they support a 16mhz and 32mhz.  If you don't see this in the boards menu and don't know how to check, you can capture the value by digging through the compiler output when you build for a line that contains "-DF_CPU=" and capture the number that follows the equal sign.
Have you selected 16Mhz and does it work?
Is this where you got your board support?  https://github.com/LGTMCU/Larduino_HSP

Makuna, 2023-01-13:
You can try this branch https://github.com/Makuna/NeoPixelBus/tree/LGT8F328P

Makuna, 2023-01-17:
@roytomeij Let me know if you need instructions on how to get a branch for testing.

roytomeij, 2023-01-17:
Apologies, for some reason, GH didn't notify me of your replies. I appreciate your speedy reply and wasn't trying to ghost you. I'll give this a try asap!

roytomeij, 2023-01-19:
> What frequency are you selecting in the boards menu? It seems they support a 16mhz and 32mhz. If you don't see this in the boards menu and don't know how to check, you can capture the value by [...]

My board package can be found here: https://github.com/dbuezas/lgt8fx (installed using PlatformIO). It runs at 32MHz by default but successfully compiles when changing the clock speed to 16MHz. I haven't yet tried running it on the actual hardware.

I've tried the branch, but it throws the same error. I must admit that I'm only 95% sure I did it correctly because this was the first time defining a specific branch in PlatformIO's ini-file. However, when looking at the source file that triggered the error, it does include the branch-specific changes, so I think I'm actually using the right branch :)

Makuna, 2023-01-19:
dig through your compiler output, did you find the `F_CPU` value?

roytomeij, 2023-01-19:
I see `DF_CPU` (`32000000L`) but not `F_CPU` ([gist with verbose build output](https://gist.github.com/roytomeij/db7bdae680b09a1050b866dd681156cc)). What am I misunderstanding?

It uses the correct branch, it seems: `NeoPixelBus @ 2.7.1+sha.e689afd` (which is the correct SHA).

Makuna, 2023-01-20:
Its a command line, D means "define", so DF_CPU is define F_CPU.
Yeah, confirmed that verbose output does define it correctly.  It also defines the correct ARCH (ARDUINO_ARCH_AVR).
The problem was the line number for error changed, I fixed one part and missed another part.  
I just pushed some fixes.

roytomeij, 2023-01-20:
It builds successfully now! I'll try to find time tonight to confirm it working as intended on the actual hardware.

Makuna, 2023-02-02:
BTW:  The board I ordered finally arrived.  While the signal is close, it's a little too fast.  I am actively working on it.

Makuna, 2023-02-02:
Ok, updated branch with latest.  Tested to work for both 800Kbps and older 400Kbps speed variants of LEDs.  I will merge this in sometime today.

Makuna, 2023-02-02:
https://github.com/Makuna/NeoPixelBus/pull/633

Makuna, 2023-02-03:
https://github.com/Makuna/NeoPixelBus/releases/tag/2.7.2

closed 2023-02-03 by Makuna