Makuna/NeoPixelBus #627: #if ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(4, 3, 0)

Andr1961, 2023-01-23:
Hi, 
ESP32 ver 1.04 example NeoPixelTest.ino
on versions older than NeoPixelBus-2.6.9 compilation error:

In file included from g:\!ya\!arduino\arduino_lib\libraries\NeoPixelBus-master\src/NeoPixelBus.h:97:0,
                 from G:\!ya\!arduino\arduino_lib\libraries\NeoPixelBus-master\examples\NeoPixelTest\NeoPixelTest.ino:14:
g:\!ya\!arduino\arduino_lib\libraries\NeoPixelBus-master\src/internal/NeoEsp32RmtMethod.h:53:43: error: missing binary operator before token "("
 #if ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(4, 3, 0)
                                           ^
exit status 1

Makuna, 2023-01-23:
NeoPixelBus v2.6.8 and before doesn't even have this line so your statement is incorrect.

Be more specific about versions.  What version of IDE (arduino?), version of board support (Ardunio Board version 1.04?), and version of IDF, and version of the library.

But older versions of the library are not directly supported.  I don't have bandwidth for it.  Often versions are published specifically due to changes in the IDF and thus can longer be backward compatible.

Andr1961, 2023-01-23:
IDE 1.8.16 
ESP32 1.0.4
SdkVersion: v3.2.3-14-gd3e562907
ERR>=NeoPixelBus-2.7 (NeoPixelBus-2.6.9 Ok)
Ok, I'll be on 2.6.9 (adafruit and Fastlend compile without errors)

closed 2023-01-23 by Andr1961

Makuna, 2023-01-23:
2.6.9+ rely on APIs that are only on IDE 4.3.0.