Makuna/NeoPixelBus #632: error: 'VARIANT_MCK' was not declared in this scope (and the same for NOT_A_PIN definition)

MNS26, 2023-02-01:
**NOTE:  If you are seeking help or have questions, this is NOT the place to do it.**  For questions and support, jump on Gitter and ask away.  
[image]

**Describe the bug**
Using this library with the RP2040 results in a build failure due to `VARIANT_MCK` and `NOT_A_PIN` not being defined

**To Reproduce**
Steps to reproduce the behavior:
1. Install library
2. create instance 
3. build for RP2040
4. error

**Expected behavior**
it completing a build

**Development environment (please complete the following information):**
 - OS: Linux
 - Build Environment: PlatformIO `Core` 6.1.6 `Home` 3.4.3
 - Board target: Rasperry Pi Pico (RP2040)
 - Library version: Latest github version

**Minimal Sketch that reproduced the problem:**
If an example will demonstrate the problem, list which example you used.
If its your sketch, take the time to remove unneeded parts and minimize it to the smallest sketch that reproduces the problem.  Then include the sketch here using `code` markup.  Start with a line with three single left quotes and then end with a line with three single left quotes. 
```cpp
#include <Arduino.h>
#include "NeoPixelBus.h"

#define PIXEL_COUNT 1;

NeoPixelBus<NeoGrbFeature, Neo800KbpsMethod> strip(PIXEL_COUNT, PIN_LED); //PIN_LED=16 defined in platfrom.ini

void setup(){}
void loop(){}
```

**Additional context**
working on a RTOS system for a RC helicopter and the adafruit library doesnt work, and found this one instead

MNS26, 2023-02-01:
```cpp
 *  Executing task in folder Pico-Heli: platformio run --environment pico 

Processing pico (platform: https://github.com/maxgerhardt/platform-raspberrypi.git; board: waveshare_rp2040_zero; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/waveshare_rp2040_zero.html
PLATFORM: Raspberry Pi RP2040 (1.7.0+sha.9a13c25) > RP2040 Zero
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, jlink, picoprobe, raspberrypi-swd)
PACKAGES: 
 - framework-arduinopico @ 1.20605.0+sha.e105c53 
 - tool-rp2040tools @ 1.0.2 
 - toolchain-rp2040-earlephilhower @ 5.100300.221223 (10.3.0)
Flash size: 2.00MB
Sketch size: 2.00MB
Filesystem size: 0.00MB
Maximium Sketch size: 2093056 EEPROM start: 0x101ff000 Filesystem start: 0x101ff000 Filesystem end: 0x101ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 38 compatible libraries
Scanning dependencies...
Dependency Graph
|-- FreeRTOS @ 1.0.0
|   |-- Adafruit TinyUSB Library @ 1.13.3
|-- MPU6050 @ 0.6.0
|   |-- Wire @ 1.0
|   |   |-- Adafruit TinyUSB Library @ 1.13.3
|-- TheDIYGuy999 SBUS @ 1.1.0+sha.dd18ed0
|-- NeoPixelBus @ 2.7.1+sha.d3bc27b
|   |-- SPI @ 1.0
|   |   |-- Adafruit TinyUSB Library @ 1.13.3
|-- Adafruit TinyUSB Library @ 1.13.3
Building in release mode
Compiling .pio/build/pico/FrameworkArduinoBootloader/boot2_w25q16jvxq_4_padded_checksum.S.o
Compiling .pio/build/pico/lib45c/Adafruit_TinyUSB_Arduino/arduino/Adafruit_TinyUSB_API.cpp.o
Compiling .pio/build/pico/lib45c/Adafruit_TinyUSB_Arduino/arduino/Adafruit_USBD_CDC.cpp.o
Compiling .pio/build/pico/lib45c/Adafruit_TinyUSB_Arduino/arduino/Adafruit_USBD_Device.cpp.o
Compiling .pio/build/pico/lib45c/Adafruit_TinyUSB_Arduino/arduino/Adafruit_USBH_Host.cpp.o
Compiling .pio/build/pico/lib45c/Adafruit_TinyUSB_Arduino/arduino/hid/Adafruit_USBD_HID.cpp.o
Compiling .pio/build/pico/lib45c/Adafruit_TinyUSB_Arduino/arduino/midi/Adafruit_USBD_MIDI.cpp.o
Compiling .pio/build/pico/lib45c/Adafruit_TinyUSB_Arduino/arduino/msc/Adafruit_USBD_MSC.cpp.o
Compiling .pio/build/pico/lib45c/Adafruit_TinyUSB_Arduino/arduino/ports/esp32/Adafruit_TinyUSB_esp32.cpp.o
Compiling .pio/build/pico/lib45c/Adafruit_TinyUSB_Arduino/arduino/ports/nrf/Adafruit_TinyUSB_nrf.cpp.o
Compiling .pio/build/pico/lib45c/Adafruit_TinyUSB_Arduino/arduino/ports/rp2040/Adafruit_TinyUSB_rp2040.cpp.o
Compiling .pio/build/pico/lib45c/Adafruit_TinyUSB_Arduino/arduino/ports/samd/Adafruit_TinyUSB_samd.cpp.o
Compiling .pio/build/pico/lib45c/Adafruit_TinyUSB_Arduino/arduino/webusb/Adafruit_USBD_WebUSB.cpp.o
Compiling .pio/build/pico/lib45c/Adafruit_TinyUSB_Arduino/class/audio/audio_device.c.o
Compiling .pio/build/pico/lib45c/Adafruit_TinyUSB_Arduino/class/bth/bth_device.c.o
Compiling .pio/build/pico/lib45c/Adafruit_TinyUSB_Arduino/class/cdc/cdc_device.c.o
Compiling .pio/build/pico/lib45c/Adafruit_TinyUSB_Arduino/class/dfu/dfu_device.c.o
Compiling .pio/build/pico/lib45c/Adafruit_TinyUSB_Arduino/class/dfu/dfu_rt_device.c.o
Compiling .pio/build/pico/lib45c/Adafruit_TinyUSB_Arduino/class/hid/hid_device.c.o
Compiling .pio/build/pico/lib45c/Adafruit_TinyUSB_Arduino/class/hid/hid_host.c.o
Compiling .pio/build/pico/lib45c/Adafruit_TinyUSB_Arduino/class/midi/midi_device.c.o
Compiling .pio/build/pico/lib45c/Adafruit_TinyUSB_Arduino/class/msc/msc_device.c.o
Compiling .pio/build/pico/lib45c/Adafruit_TinyUSB_Arduino/class/net/ecm_rndis_device.c.o
Compiling .pio/build/pico/lib45c/Adafruit_TinyUSB_Arduino/class/net/ncm_device.c.o
Compiling .pio/build/pico/lib45c/Adafruit_TinyUSB_Arduino/class/usbtmc/usbtmc_device.c.o
Compiling .pio/build/pico/lib45c/Adafruit_TinyUSB_Arduino/class/vendor/vendor_device.c.o
Compiling .pio/build/pico/lib45c/Adafruit_TinyUSB_Arduino/common/tusb_fifo.c.o
Compiling .pio/build/pico/lib45c/Adafruit_TinyUSB_Arduino/device/usbd.c.o
Compiling .pio/build/pico/lib45c/Adafruit_TinyUSB_Arduino/device/usbd_control.c.o
Compiling .pio/build/pico/lib45c/Adafruit_TinyUSB_Arduino/host/hub.c.o
Compiling .pio/build/pico/lib45c/Adafruit_TinyUSB_Arduino/host/usbh.c.o
Compiling .pio/build/pico/lib45c/Adafruit_TinyUSB_Arduino/portable/chipidea/ci_hs/dcd_ci_hs.c.o
Compiling .pio/build/pico/lib45c/Adafruit_TinyUSB_Arduino/portable/chipidea/ci_hs/hcd_ci_hs.c.o
Compiling .pio/build/pico/lib45c/Adafruit_TinyUSB_Arduino/portable/espressif/esp32sx/dcd_esp32sx.c.o
Compiling .pio/build/pico/lib45c/Adafruit_TinyUSB_Arduino/portable/microchip/samd/dcd_samd.c.o
Compiling .pio/build/pico/lib45c/Adafruit_TinyUSB_Arduino/portable/nordic/nrf5x/dcd_nrf5x.c.o
[... 506 lines cut ...]
  818 |             *timeReset = TC_CCR_CLKEN | TC_CCR_SWTRG;
      |                          ^~~~~~~~~~~~
.pio/libdeps/pico/NeoPixelBus/src/internal/NeoArmMethod.h:818:41: error: 'TC_CCR_SWTRG' was not declared in this scope
  818 |             *timeReset = TC_CCR_CLKEN | TC_CCR_SWTRG;
      |                                         ^~~~~~~~~~~~
.pio/libdeps/pico/NeoPixelBus/src/internal/NeoArmMethod.h:838:9: error: there are no arguments to 'TC_Stop' that depend on a template parameter, so a declaration of 'TC_Stop' must be available [-fpermissive]
  838 |         TC_Stop(TC1, 0);
      |         ^~~~~~~
In file included from src/main.cpp:6:
[... cut ...]
=============================================================== [FAILED] Took 6.03 seconds ===============================================================

Environment    Status    Duration
-------------  --------  ------------
pico           FAILED    00:00:06.032
========================================================= 1 failed, 0 succeeded in 00:00:06.032 =========================================================

 *  The terminal process "platformio 'run', '--environment', 'pico'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 

```

Makuna, 2023-02-01:
Duplicate of https://github.com/Makuna/NeoPixelBus/issues/578, this board is currently not supported, see the primary issue.

closed 2023-02-01 by Makuna