Makuna/NeoPixelBus #638: Using NeoRgbwUcs8904Feature with NeoPixelBrightnessBus causes heap corruption

blazoncek, 2023-02-04:
**Describe the bug**
Reporting this on behalf of [WLED](https://github.com/Aircoookie/WLED) project where we want to add support for UCS890x chipsets. Even though WLED does not yet use 16-bit internal logic we adjusted output to be aware of 16-bit UCS LED chip.

UCS8903 seems to work correctly (using `NeoRgbwUcs8903Feature`) but as soon as we create a `NeoPixelBrightnessBus` object with `NeoRgbwUcs8904Feature` heap gets corrupted and abort is called causing ESP32 to reboot. Technically it (abort()) can happen on another core, not within NeoPixelBus (or NeoPixelBrightnessBus).

**To Reproduce**
WLED allows dynamic selection of LED types so we use operator new to create appropriate bus.
The logic goes like so:
```c++
void *busptr = new NeoPixelBrightnessBus<NeoRgbwUcs8904Feature, NeoEsp32RmtNWs2812xMethod>(len, pin, channel);
(static_cast<NeoPixelBrightnessBus<NeoRgbwUcs8904Feature, NeoEsp32RmtNWs2812xMethod>*>(busptr))->Begin();
...
```
We use similar approach for all other features and methods and so far none caused heap corruption.

**Expected behavior**
No heap corruption.

**Development environment (please complete the following information):**
 - OS: [macOS 10.14]
 - Build Environment [MS VSC+PIO]
 - Board target [Espressif ESP32]
 - Library version [v2.6.9]

**Minimal Sketch that reproduced the problem:**
See above.

**Additional context**
There is no need to call any other NeoPixelBrightnessBus method as heap gets corrupted within Begin() method (or so it seems). I have not yet had the time to debug NeoPixelBus internals since it is beyond my expertise.

blazoncek, 2023-02-04:
Update: Changing to v2.7.1 does not solve the issue. v2.7.2 will not compile with PlatformIO ESP32 core 3.5.0 (which we use for ESP32). Compilation fails for DotStar SPI methods.

Sample crash:
```
PIN ALLOC: Pin 3 successfully allocated by 0x82 (130)
Creating bus.  <-- entering function with new NeoPixelBrightnessBus<...>
Starting begin.  <-- new completed, entering WLED begin(); calling Begin()
Begin done.  <-- Begin() called
Bus created.  <-- returned from WLED begin()
CORRUPT HEAP: multi_heap.c:432 detected at 0x3ffb9a18
abort() was called at PC 0x4008f897 on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4008a758:0x3ffb1d60 0x4008a9d5:0x3ffb1d80 0x4008f897:0x3ffb1da0 0x4008fe5c:0x3ffb1dc0 0x4008351d:0x3ffb1de0 0x4008354e:0x3ffb1e00 0x40088771:0x3ffb1e20 0x4000beaf:0x3ffb1e40 0x40136a5c:0x3ffb1e60 0x400ee0c1:0x3ffb1f00 0x40118eb5:0x3ffb1f50 0x401191a2:0x3ffb1f90 0x4013af05:0x3ffb1fb0 0x4008c4da:0x3ffb1fd0
  #0  0x4008a758:0x3ffb1d60 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #1  0x4008a9d5:0x3ffb1d80 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #2  0x4008f897:0x3ffb1da0 in multi_heap_assert at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c:380
      (inlined by) multi_heap_malloc_impl at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c:432
  #3  0x4008fe5c:0x3ffb1dc0 in multi_heap_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:321
  #4  0x4008351d:0x3ffb1de0 in heap_caps_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:232
  #5  0x4008354e:0x3ffb1e00 in heap_caps_malloc_default at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:232
  #6  0x40088771:0x3ffb1e20 in _malloc_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/syscalls.c:37
  #7  0x4000beaf:0x3ffb1e40 in ?? ??:0
  #8  0x40136a5c:0x3ffb1e60 in Print::printf(char const*, ...) at /Users/blaz/.platformio/packages/framework-arduinoespressif32@3.10006.210326/cores/esp32/Print.cpp:254
  #9  0x400ee0c1:0x3ffb1f00 in BusDigital::BusDigital(BusConfig&, unsigned char, ColorOrderMap const&) at wled00/FX_fcn.cpp:1384
      (inlined by) BusManager::add(BusConfig&) at wled00/bus_manager.h:710
  #10 0x40118eb5:0x3ffb1f50 in WLED::loop() at wled00/wled.cpp:170
  #11 0x401191a2:0x3ffb1f90 in loop() at wled00/wled00.ino:20
  #12 0x4013af05:0x3ffb1fb0 in loopTask(void*) at /Users/blaz/.platformio/packages/framework-arduinoespressif32@3.10006.210326/cores/esp32/main.cpp:23
  #13 0x4008c4da:0x3ffb1fd0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8

---WLED 0.14.0-b2 2301290 INIT---
esp32 v3.3.5-1-g85c43024c
arduino-esp32 v1.0.x

CPU:   ESP32-D0WDQ6 rev.1, 2 core(s), 240MHz.
FLASH: 4MB, Mode 2 (DIO), speed 40MHz.
heap 279564
...
```

Makuna, 2023-02-04:
@blazoncek "v2.7.2 will not compile with PlatformIO ESP32 core 3.5.0 (which we use for ESP32). Compilation fails for DotStar SPI methods."
I do not specifically support PlatformIO, so any compile issues will need to be diagnosed by someone who does.  Please create another issue for that separate topic.

blazoncek, 2023-02-04:
I will.
I tested with 2.7.2 on ESP32-C3 but the problem persists.

blazoncek, 2023-02-04:
BTW the compile failure is due to DotStarEsp32DmaSpiMethod having new buscfg struct members (spi_common.h changed) but that is irrelevant for this issue.

Makuna, 2023-02-04:
found an issue, merged in some fixes to try
https://github.com/Makuna/NeoPixelBus/pull/639

blazoncek, 2023-02-04:
I dared to patch 2.7.1 with NeoColorFeatures.h and can report success (as I am unable to use 2.7.2 with ESP32).
Is it possible to backport this to 2.6.9?

Makuna, 2023-02-04:
Is not being able to use 2.7.2 just due to the SPI compile issue?
I just can't support older versions and would rather spend efforts on fixing issues.

Makuna, 2023-02-04:
(leave open, I will close when the issue gets released into a version)

Makuna, 2023-02-04:
https://github.com/Makuna/NeoPixelBus/issues/641
Also fixed

blazoncek, 2023-02-04:
Thank you.
It looks like #607 broke compile of SPI on older espressif32 platforms. And there are myriad warnings issued when compiling 2.7.1 and up regarding [[maybe_unused]] compiler directive.

blazoncek, 2023-02-04:
It was a pain (due to [[maybe_unused]]) but latest commit compiles under espressif32@3.5.0 (IDF 4.3.x)

Makuna, 2023-02-04:
GNU C++ version compatibility is the reason of [[maybe_unused]].  Sometimes a newer compiler version is used and sometimes they provide a flag to say use old rules.  So this maybe fixed on your side by uping the flag.

I really don't want to backtrack on using it since its a feature that has been around for sometime in the compiler.

blazoncek, 2023-02-04:
Agreed.
I just commented as I wanted minimal amount of changes for testing.

Makuna, 2023-02-10:
v2.7.3

closed 2023-02-10 by Makuna