Makuna/NeoPixelBus #649: Compilation error for ESP8266 using PlatformIO ESP8266 platform 4.1.0 (latest)

blazoncek, 2023-02-11:
**Describe the bug**
When compiling for ESP8266 using PlatformIO (with latest platform v4.1.0) errors are displayed and compile fails.

```
.pio/libdeps/d1_mini_custom_debug/NeoPixelBus/src/internal/NeoEsp8266DmaMethod.h:214:29: error: expected unqualified-id before '[' token
     NeoEsp8266DmaMethodBase([[maybe_unused]] uint8_t pin, uint16_t pixelCount, size_t elementSize, size_t settingsSize) : 
                             ^
.pio/libdeps/d1_mini_custom_debug/NeoPixelBus/src/internal/NeoEsp8266DmaMethod.h:214:29: error: expected ')' before '[' token

.pio/libdeps/d1_mini_custom_debug/NeoPixelBus/src/internal/NeoEsp8266I2sDmx512Method.h:161:35: error: expected unqualified-id before '[' token
     NeoEsp8266I2sDmx512MethodBase([[maybe_unused]] uint8_t pin, uint16_t pixelCount, size_t elementSize, size_t settingsSize) : 
                                   ^
.pio/libdeps/d1_mini_custom_debug/NeoPixelBus/src/internal/NeoEsp8266I2sDmx512Method.h:161:35: error: expected ')' before '[' token
```

The problem lies in NeoEsp8266DmaMethod.h and NeoEsp8266I2sDmx512Method.h with compiler directive `[[maybe_unused]]` which does not compile for some odd reason.
Commenting out those 2 directives resolves the issue.

**To Reproduce**
try to compile sample code

**Expected behavior**
No errors during compilation.

**Development environment (please complete the following information):**
 - OS: [macOS 10.14]
 - Build Environment [PlatformIO 6.1.6]
 - Board target [ESP8266 v4.1.0]
 - Library version [v2.7.3]

**Minimal Sketch that reproduced the problem:**
```c++
NeoPixelBrightnessBus<NeoGrbFeature, NeoEsp8266Dma800KbpsMethod> *busPtr = new NeoPixelBrightnessBus<NeoGrbFeature, NeoEsp8266Dma800KbpsMethod>(len, pin);
```

**Additional context**
Above sample code from WLED bus_wrapper.h.

softhack007, 2023-02-11:
At least in the "gcc world", there would be an alternative: 
`__attribute__((unused)) ` or `[[gnu::unused]]`

https://gcc.gnu.org/onlinedocs/gcc-10.4.0/gcc/Attribute-Syntax.html#Attribute-Syntax

Another option could be to create a macro like MAYBE_UNUSED, which is either empty, or defined as the keyword that is supported by the compiler.

Makuna, 2023-02-11:
GCC does support [[maybe_unused]].  

I believe this is due to the older compiler you are using or a compile flag that states to use older rules. 
The latest Arduino board support for ESP8266 does not have this issue.   (-std=gnu17)

Why are you not trying to fix your build environment?

blazoncek, 2023-02-11:
I am using PlatformIO 6.1.6 (+VSC) using supplied ESP8266 platform 4.1.0 using the following build flags:
```
build_flags =
  -DMQTT_MAX_PACKET_SIZE=1024
  -DSECURE_CLIENT=SECURE_CLIENT_BEARSSL
  -DBEARSSL_SSL_BASIC
  -D CORE_DEBUG_LEVEL=0
  -D NDEBUG
  #build_flags for the IRremoteESP8266 library (enabled decoders have to appear here)
  -D _IR_ENABLE_DEFAULT_=false
  -D DECODE_HASH=true
  -D DECODE_NEC=true
  -D DECODE_SONY=true
  -D DECODE_SAMSUNG=true
  -D DECODE_LG=true
  -DWLED_USE_MY_CONFIG
  -DESP8266
  -DFP_IN_IROM
  -DPIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_190703
  -DPIO_FRAMEWORK_ARDUINO_LWIP_HIGHER_BANDWIDTH
  -DVTABLES_IN_FLASH
  -DMIMETYPE_MINIMAL
```
So no special compiler specific build flags and it will not compile.

I would appreciate any hint on how to fix it myself (other than forking and removing `[[maybe_unused]]`). I did not manipulate default installation in any way.

Makuna, 2023-02-11:
try adding

-std=gnu17

this is what the Arduino builds use.

blazoncek, 2023-02-11:
Thank you. Will try it ASAP.

Makuna, 2023-02-11:
also found a similiar flag

-std=gnu++17

xavor92, 2023-02-17:
Hey,

since I searched way too long to "just use c17", here a link for whoever else struggles:

https://community.platformio.org/t/how-to-use-c-17-with-esp8266/18950/6

blazoncek, 2023-03-12:
I have now thoroughly tested compiling with PlatformIO `platform = espressif8266@4.1.0` with
```ini
platform_packages = platformio/framework-arduinoespressif8266 ;latest
                    platformio/toolchain-xtensa ;@2.100300.220621
                    platformio/tool-esptool
                    platformio/tool-esptoolpy
```
And can report that compiling NeoPixelBus v2.7.0 or above compiles without major issues.
I am closing this issue as it appears latest platform and toolchain resolves compiling.

closed 2023-03-12 by blazoncek