Makuna/NeoPixelBus #653: Flickering due to NeoEsp8266DmaSpeed800KbpsBase running too fast at 833kHz

wangnick, 2023-02-19:
I drive a strip of 176 WS2812B using an ESP8266 (plus an SN74AHCT125 to level shift the GPIO3 output) and the NeoEsp8266DmaWs2812xMethod of the NeoPixelBus library. Power supply is sound. 

The LEDs sit rather packed in a dome and therefore become reasonably warm. When reaching about 70°C I observe more and more flickering towards the end of the strip.

Examinig the issue in more details with a logic analyser recording the bit stream towards LED 1, 152, 172, 173,174 and 175, I see holes introduced stochastically into the bit stream, i.e. sometimes the data line stays low for a full bit period.

I managed to catch such a hole being introduced by LED 173 at DO. This occurred when LED 173 was fed at DI with a bit only 992ns long (a regular 424ns high level followed by a low level of only 568ns) followed by a bit of 1400ns (a regular 424ns high level followed by a 976ns low level); in this case LED 173 swallowed the second one of these two strange bits. See attached recording:
[image]

What I also observed was that the input datastream to LED 1 was running at 833kHz instead of 800kHz, with every bit lasting 1200μs instead of 1250μs. This gave me the suspicion that the flickering might be due to the WS2812B trying to perform some form of frequency recovery, and failing this due to the input stream being at too high a pace.

Checking the library source I found the following code:
```
class NeoEsp8266DmaSpeed800KbpsBase : public NeoEsp8266DmaSpeedBase
{
public:
    const static uint32_t I2sClockDivisor = 3; // 0-63
    const static uint32_t I2sBaseClockDivisor = 16; // 0-63
    const static uint32_t ByteSendTimeUs = 10; // us it takes to send a single pixel element at 800khz speed
};
```

Now, I understand that the I2S clocking is derived on the ESP8266 from a base clock running at 160MHz, which divided by the I2sBaseClockDivisor and then the I2sClockDivisor gives the I2S bit clock, and that every WS2812B bit is generated by four I2S bits (either 0b1000 or 0b1110), and that therefore the resulting frequency is indeed 160MHz/16/3/4 = 833kHz.

I therefore changed the source code to:
```
    const static uint32_t I2sClockDivisor = 5; // 0-63
    const static uint32_t I2sBaseClockDivisor = 10; // 0-63
```
and guess what: The generated bit stream sits tight at 800kHz, and all flickering is gone!

Kind regards,
Sebastian

Makuna, 2023-02-19:
Are you actually capturing at the LED input for those further down the strip ?  Or are you just capturing the complete output at the first LED?  Note that the LEDs just don't pass through the signal, they actually reform it and correct it.  You can see this if you feed a 5v LED with a signal that is 3.3v, the output is now 5v.  Further, it also reforms the output (using a scope you may see spikes on the input but the output is clean).

Do you know exactly which strip/chips you have?  WS2812b (from v1.4 spec sheet) should be able to handle pulse widths from 800ns to 1380ns.  With highs for zero 220-380 and highs for ones 580-1000.  The companies are notorious for changing chip specs without changing model numbers; but it's usually just the reset time that is the problem.

You are getting close to the maximum operating temperature of these LEDs.  My spec states -25～+85C for operating range.  If this happens only when hot like you state, I think you are hitting a temperature issue.

The clock reference is not the CPU clock; It's the peripheral clock.  You can vary the CPU clock in the tools menu but the peripheral clock will stay the same.  But your math looks good.

Note that a faster speed may have been selected due to slight variance of i2s output made it not compatible in the maximum high pulse of the zero bit (380ns). so, to get this closer in spec the speed was increased overall but still within spec.

I will put it under the probes and see if I see the same things as you.

Makuna, 2023-02-19:
I just noticed something on their website for the latest LEDs as of 2019.
"Signal recognition range reduced to less than 2.8V." 
So, the level shifters shouldn't be required anymore unless the wire run is long enough to drop the signal voltage below 3v.  You can tell if you have the newer chips by looking at the circuit, if there are no other components than the LEDS (surface mount cap and or resistors) then you have the new ones that shouldn't need the level shifters.

wangnick, 2023-02-19:
> Are you actually capturing at the LED input for those further down the strip ? Or are you just capturing the complete output at the first LED?

I captured in one go the digital bit stream at 125MS/s at DI of LED 1, at DI of LED 152, at DI of LED 172, at DI of LED 173, at DI of LED 174 and at DI of LED 175, and the analog bit stream at DI of LED 173 at 50MS/s. Attached the recording of the event I describe above, in Saleae Logic 2 format: [neopixelbusflicker.singleevent.sal.zip](https://github.com/Makuna/NeoPixelBus/files/10777651/neopixelbusflicker.singleevent.sal.zip)

> Do you know exactly which strip/chips you have?
> I just noticed something on their website for the latest LEDs as of 2019.
> "Signal recognition range reduced to less than 2.8V."

Unfortunately not. In fact I might have bought them back in 2016. They look like this: [image]

I do not think that the SN74AHCT125 would introduce noticeable jitter.

> You are getting close to the maximum operating temperature of these LEDs. My spec states -25～+85C for operating range. If this happens only when hot like you state, I think you are hitting a temperature issue.

It happens already as from 70°C (checked with both an infrared probe and a Flir One Pro). And with the timings adapted to be exactly 800kHz the strip is rock stable into the 90°C.

It occurs to me, though, that this might not be a true "bug" of the library. After all, one can easily adapt the timings to ones liking or necessities in an own class inheriting from NeoEsp8266DmaSpeedBase.

Kind regards,
Sebastian

Makuna, 2023-02-19:
yeah, those are the older ones (you can see the extra cap next to it).  So the level shifter is probably still required.

Yes, you can always provide modified classes to build up your own, it's part of the power of the template design of the library.  I will still take a look and might adjust it.  Things do change in the core causing old design work arounds to get in the way.

It's interesting that it does seem that the LEDs are stalling on the data later in the strip.  Never knew this could happen BUT there have been some unexplained issues like this before.  Temperature does change timing oscillators; so if the ESP8266 is very warm it may run faster/slower than expected.  The same with the LEDs tiny processors.  With the data stream already faster but within spec the two together could be causing this (and some of the other undiagnosed flickering).

Makuna, 2023-02-22:
confirmed what you are seeing.  I will create a pull shortly.

Makuna, 2023-02-22:
Could you try a test?  What happens in your high heat environment with the values that are slower, like ...
```
    const static uint32_t I2sClockDivisor = 6; // 0-63
    const static uint32_t I2sBaseClockDivisor = 9; // 0-63
```

Makuna, 2023-02-22:
https://github.com/Makuna/NeoPixelBus/pull/654
I will close when the next official release is published.

wangnick, 2023-02-23:
> Could you try a test? What happens in your high heat environment with the values that are slower, like ...
> ```
>     const static uint32_t I2sClockDivisor = 6; // 0-63
>     const static uint32_t I2sBaseClockDivisor = 9; // 0-63
> ```

Now trying ...

Makuna, 2023-02-23:
thanks, good information to have.

closed 2023-04-11 by Makuna