Makuna/NeoPixelBus #655: How[DELETE ME]

jumper-d-1981, 2023-02-23:
### STOP
If you are seeking support, then please use the Discussions feature or gitter channel by following one of these links.  
[NeoPixelBus Discussions](https://github.com/Makuna/NeoPixelBus/discussions)  
[NeoPixelBus Gitter Channel](https://gitter.im/Makuna/NeoPixelBus)  

If you submit issues that are not traceable bugs or feature requests, it will just get closed.

Makuna, 2023-02-23:
You found the discussions, deleting. ;-)

I am curious as to your experience that you didn't notice the references to Discussions?  You did see them?  I get this a lot and I have been asking GitHub for a solution to promote Discussions for Q/A rather than Issues and having your experience documented might shed some light on a better approach/solution.

jumper-d-1981, 2023-02-24:
Well, to be honest, very few people have active discussions on their github pages...at least of the ones I have viewed.  So my 1st reaction is to use the "Issues" as the forum.  Seems to typically work.  BUT...now I know.  lol.

closed 2023-02-24 by Makuna