Makuna/NeoPixelBus #669: Add Support for TM1906/FW1906

Evanvdberg93, 2023-03-26:
Following some earlier discussions in the discord, I’d like to request support for the FW1906/TM1906 be added.

Makuna, 2023-03-26:
Interesting that I have the spec sheet (in chinese) but I didn't see support for it yet.  I did see TM1814 & TM1914 but they are not compatible.  I will take a look.

Makuna, 2023-03-27:
Do you actually have some of these?  
**If you don't**, I would avoid them if you can.  The first color element of pixel can not be 255, not that this complete loss, it's just annoying as they use it as a flag to set the constant current brightness.  Further, if you are not directly using my library (like using WLED or others) it is unlikely to work well as it will REQUIRE that you set the pixel current once (a call in my library) that is not common to other NeoPixel like LEDs; and the libraries that use my code often don't support this rare call well. 

**If you do**, you will have to be the tester of the changes as I don't have any and very unlikely to purchase any.  I will need to know the order of the color elements RGB versus GRB and if your strip is driving 4 pixels per chip (5v) or only one pixel per chip (12v?).  They can be wired in all sorts of ways.

Evanvdberg93, 2023-03-27:
I don’t have some of them no. I was looking at getting some since as far as I’m aware they’re the first individually addressable RGBCCT strips, but support for them on the library side seems to be nonexistent.

interesting what you mention w.r.t the constant current flag, and the downstream libraries issues. 

In terms of platform, ESP32 though I’m open to switching if it makes support easier, as at this point these strips seem to be quite the hassle to even find a supporting library.

Makuna, 2023-03-27:
The TM1814 is similar.  There is a NeoTm1814Settings that you instance and make calls to set the current need for each of the color elements and then call `strip.SetPixelSettings(settings)` with it.  This would mean you need to know what the current requirements are for the RGB and the white LEDs (and how they are attached).  CCT means color coordinated temperature, it doesn't define how the LEDs are ordered or mixed with the RGB.

marschr, 2023-05-07:
Hi. I have two FW1906 strips. They work fine with a modified version of FastLED on a Arduino mega board I had around.
It uses one IC controller for each 24 volt sub-segment with six LED-ICs each. The controller ICs have six output channels and only the first five are used. I need to reinstall my arduino to check the color channel ordering for sure, but it was someting like R,G,B,WW,CW,empty.
My naive hacky approach on FastLED was to modify another WS2812B code by filling all the six color channels with the int8 values for each.
Recently I started digging into WLED code trying to make it compatible with it, then I saw how much work NeoPixelBus was doing behing the scenes and arrived here. Any tip on how to proceed? If I'm understanding the architecture correctly, should I add a custom NeoPixel Feature, like the one for TM1814, but with all the six channel components?
Can one derive the current from the nominal power consumption from the spec sheet or should it be measured?
Is this a good place to test it, should I join a discord or something?

Makuna, 2023-05-07:
TM1906 Spec states "each chip has 4 sets of RGB"; or 12 PWM outputs.  This sounds like the FW1906 is different.  I just searched around and yes, they are not the same.  I still don't have a spec for it though.

created a [separate issue](https://github.com/Makuna/NeoPixelBus/issues/702) for it

BLeibbrand, 2023-05-24:
I try to use the FW1906-RGBCCT-84x12 with the WT32-ETH01 as the WLED controller. So if support is added that would be perfect.

BLeibbrand, 2023-06-04:
In the https://www.reddit.com/r/WLED/comments/11hdf56/is_fw1906_compatible_with_wled/?onboarding_redirect=%2Fr%2FWLED%2Fcomments%2F11hdf56%2Fis_fw1906_compatible_with_wled%2F is more detail.

https://github.com/Makuna/NeoPixelBus/issues/702

https://gree-leds.com/ic-protocol/fw1906-ic-protocol-and-datasheet.html

https://www.artleds.com/blog/fw1906-ic-pixel-protocol-overview

I tested the hardware using a single data pin type in WLED. Using the calculation: 84 LED/m on a 5m strip but with 6 LED on one "double" controller:

(84x5)/6= 70 which should be correct IF the chip was handling only RGB.

So I had to use the double amount of 140 to get all the LEDs touched. The data is shifted through the chain, but the result is RGB 1 on a block and RGB 2 on the other block of LEDS (6LEDS/one "double" controller)

Makuna, 2023-06-05:
@BLeibbrand Note the title of this issue, it's about TM1906 not the FW1906.   They are very different chips.
 I mentioned above that I separated the issue for FW1906 and you seem to have already found and commented on it.