Makuna/NeoPixelBus #684: NeoPixelBus on a EPS32 board with esp-idf framework

roykrikke, 2023-04-10:
**Is your feature request related to a problem? Please describe.**
This feature request is not based on a problem, but missing support for esp-idf framework.

**Describe the solution you'd like**
I'd like it if NeoPixelBus could support a EPS32 board with the end goal to have an integration possibly inside ESPHome via the esp-idf framework
Board example e.g. like [Atom Lite](https://docs.m5stack.com/en/core/atom_lite)

**Additional context**
There are several [ESP32 based boards](https://docs.m5stack.com/en/products) designed which could be used with NeoPixelBus for controlling e.g. SK6812 or WS2812 led strips, it would be great if they could be run inside ESPHome via NeoPixelBus.

Makuna, 2023-04-10:
Have you tried it?  I would imagine with little effort on your side you should be able to just include NeoPixelBus.h and it should work.  A simple Arduino.h file containing the very few specific arduino calls made and I would think it would work.  

It is unlikely that I would make the effort but I will mark it and leave it active.

roykrikke, 2023-04-10:
> Have you tried it? I would imagine with little effort on your side you should be able to just include NeoPixelBus.h and it should work. A simple Arduino.h file containing the very few specific arduino calls made and I would think it would work.
> 
> It is unlikely that I would make the effort without an incitive, but I will mark it and leave it active. I have tons of features to work as is.

I know C and C++ but I don't quite understand what you mean with your explanation. Can you help me on my way so that I can make the effort myself?

Makuna, 2023-04-10:
My library includes "Arduino.h", the idf will not have this, so create an empty one in your project.  As you compile and find errors about missing arduino specific calls/defines, add wrappers in your Arduino.h as needed.  I suspect there is just going to be a few.  digitalWrite(), pinMode() are examples.  Heck, you might be able to just copy them from the ardunio esp32 source code.  Most of the lower level stuff I use is IDF calls.

Surgikill, 2023-06-09:
Just chiming in, I am also running into this issue. I'm not sure how I can go about adding "Arduino.h" into esphome.

Makuna, 2023-06-09:
I also don't know esphome.  Have you asked there if there is a way to include Arduino libraries?

@Surgikill In the compiler output, what errors are you getting?  Also include the make line that runs the compiler so I can view the defines (-Dblahblah).

Surgikill, 2023-06-09:
@Makuna This is the output when I try to compile. 

`INFO ESPHome 2023.5.5
INFO Reading configuration /config/esphome/office-presencebox.yaml...
Failed config

light.neopixelbus: [source /config/esphome/office-presencebox.yaml:260]
  
  This feature is only available with frameworks ['arduino'].
  platform: neopixelbus
  type: GRB
  variant: WS2811
  pin: GPIO23
  num_leds: 60
  name: NeoPixel Light
`

I don't even know what the make line is. ESPhome just uses yaml configurations to then flash devices.

Makuna, 2023-06-09:
ESPHome requires the framework Arduino for their use of my Library it looks like.  Again, not a ESPHome person, so talking with them first is the way to go.

trombik, 2023-07-12:
> A simple Arduino.h file containing the very few specific arduino calls made and I would think it would work.

it turned out you are right. Some examples are running fine on my board. one needs to copy some files, including many HAL files for different chips, with some modifications. Modifications to `NeoPixelBus` is minimal; skipping `SerialDumpTables()` and SPI-based driver (they are tied with hardware). It might possible to implement a simple wrapper of `esp-idf` SPI Master Driver as an alternative of arduino `SPI` library.

> It is unlikely that I would make the effort without an incitive, but I will mark it and leave it active.

do you mean, if someone maintains `esp-idf` part in the tree, you would merge the changes?

edwardtfn, 2024-06-12:
I can also help with esp-idf if you are willing to merge it.
And as soon you merge, I can also submit a pr to change ESPHome for supporting esp-idf with this library.

Makuna, 2024-06-14:
If they don't affect the Arduino building, then yes, I will merge.

edwardtfn, 2024-06-17:
To which branch should I submit the PRs?

Makuna, 2024-06-18:
> To which branch should I submit the PRs?

Master.  Master is always the next upcoming release.  Each release gets a tag into master when it was stable.  Other branches are usually prototype changes or changes so dramatic they require some time to "bake" with concerned users before they get rolled into master.  
If the changes are dramatic, they may require a branch.  But work in your clone of this repo to get things working.  Test to make sure that it compiles for Arduino.

w531t4, 2025-07-20:
voicing my interest in making this function with esp-idf.