Makuna/NeoPixelBus #700: ColorFeatures using 16bit color elements has endian issue with buffer copies to color objects

Makuna, 2023-05-06:
**Describe the bug**
Colors can have bytes swapped in their elements in the output stream to the NeoPixels depending on the endianess of the uC.

**To Reproduce**
Use a color feature with 16 bit elements like
```
NeoPixelBus<NeoGrbWs2816Feature, NeoEsp32Rmt1Ws2816Method> strip(1, TX);
```
set the color object with the elements set to 0x0001.
note output on pin has order 0x0100

**Expected behavior**
Bytes not swapped.

**Additional context**
https://github.com/Makuna/NeoPixelBus/discussions/698

Makuna, 2023-05-06:
https://github.com/Makuna/NeoPixelBus/pull/699

tablatronix, 2023-05-07:
Looks like there is something wrong with the third byte

color 65535

[image]

tablatronix, 2023-05-07:
*p++ = color.G >> 8;
        *p++ = color.G & 0xff; // ?

Makuna, 2023-05-07:
sigh, trusted my unit test too much, it only had values in the low nibble of the bytes.

https://github.com/Makuna/NeoPixelBus/pull/701

Makuna, 2023-06-27:
https://github.com/Makuna/NeoPixelBus/releases/tag/2.7.6

closed 2023-06-27 by Makuna