Makuna/NeoPixelBus #702: Support FW1906 LED driver chip

Makuna, 2023-05-07:
**Is your feature request related to a problem? Please describe.**
The FW1906 is often used in strips with a single package LEDs of RGB-WWx configuration.

**Describe the solution you'd like**
It seems to be a fully WS2812x signal compatible chip (need spec) driving six channels rather than just three.
`NeoRgbwwxFeature` similar to the already existing `NeoRgbwxxFeature` but exposing the second white channel?
RgbwwColor to match it?

**Describe alternatives you've considered**
Stick with the NeoRgbFeature/RgbColor; rely on the sketch writer to know that even pixels are the WWx rather than RGB.

**Additional context**

Makuna, 2023-05-07:
@marschr Note the alternatives listed above.  You can do this today with the caveat that even pixels the RGB represents WWx in your case.

marschr, 2023-05-08:
I made some progress today, I got my ESP32 working with `NeoPixelBus<NeoRgbwwxFeature, NeoEsp32I2s0800KbpsMethod>`, `NeoRgbwwxFeature` being a modified version of `NeoRgbwxxFeature`. I can set the strip to the colors accordingly with `SetPixelColor()`. `ClearTo()` seems to have some issues with my `RgbwwColor`, its sorta sampling all the 5 color channels and outputting a mean value to all of them, I need to look further into it and iron things out. Also, color channel ordering is G, R, B, CW, WW.

> **Additional context** Is this chip used on a strip that has other configurations, like two RGB LEDs or even RGBWWW?

I haven't seen it being used in other LED strips, seemed to a very application specific LED controller IC, it doesn't even have any laser markings on the packaging, I have a chinese datasheet for it if needed. Now for these single package RGBWW LEDs I couldn't find any spec.

Makuna, 2023-05-08:
Modify your Feature to correct order, NeoGrbwwxFeature.  Its ok to only support the order that you need now so no need to keep NeoRgbwwxFeature.
Go ahead and create a pull request with what you have, I won't merge it, but I can take a look and provide feedback through the review system.

Makuna, 2023-05-09:
The only thing that seems broken was the count should be 5 not 4 as noted in the code review on the pull request.

Paalap, 2023-05-20:
Hi, the development is highly appreciated but it is not really working for me - it seems the color pixels are shifted I would assume I need something like "CxGRBW" instead of "RGBCWx"

To be honest I'm not fully into the project and didn't figured out how to set empty bit in between other colors

Makuna, 2023-05-21:
A quick way to test color order is to do something like this...
Use this definition `NeoPixelBus<NeoRgbFeature, NeoWs2812xMethod>' 
Then use this sample to set the colors..
```
strip.SetPixelColor(0, RgbColor(255,0,0));
strip.SetPixelColor(0, RgbColor(0,0,0));

strip.SetPixelColor(1, RgbColor(0,255,0));
strip.SetPixelColor(1, RgbColor(0,0,0));

strip.SetPixelColor(2, RgbColor(0,0,255));
strip.SetPixelColor(2, RgbColor(0,0,0));

strip.SetPixelColor(3, RgbColor(0,0,0));
strip.SetPixelColor(3, RgbColor(255,0,0));

strip.SetPixelColor(4, RgbColor(0,0,0));
strip.SetPixelColor(4, RgbColor(0,255,0));

strip.SetPixelColor(5, RgbColor(0,0,0));
strip.SetPixelColor(5, RgbColor(0,0,255));

strip.Show();
```

Then note the pixels that are lit and the colors in order.

Paalap, 2023-05-22:
the result is GRBCWx

Makuna, 2023-05-22:
@Paalap To be precise, is this color order (GRBCWx) on the strip what you observed or is this the derived order of the feature that you need?

BLeibbrand, 2023-05-24:
https://www.artleds.com/blog/fw1906-ic-pixel-protocol-overview

BLeibbrand, 2023-05-24:
https://gree-leds.com/ic-protocol/fw1906-ic-protocol-and-datasheet.html

Paalap, 2023-05-31:
Finally got some time to test - it is working, I've found my issue.

To be more detailed:
- Defined for me NeoGrbcwxFeature
- strip->SetPixelColoris working fine for me
- strip->ClearTo is creating some wired results ... is it caused by 6 vs 5 pixels?

Everything else is untested or for me it is done outside of the library

marschr, 2023-05-31:
@Paalap I think I could give some hints on why ClearTo() is not working, I did change some stuff for it to work, take a look into the `Neo6Byte5xElements.h` file here: https://github.com/marschr/NeoPixelBus/commit/854e4f891eaa554be3912a09c8a910caccc307f8 .
I borrowed it from other `features` implementation although I'm not sure if it's the most efficient way to rewrite all the addresses for all the leds.

(I couldn't find time to work on the FW1906 implementation lately and needed to buy new esp32s to test it)

Makuna, 2023-05-31:
I didn't spot anything specific that would cause ClearTo to not work.  I commented in the pulls on some minor things you will want to fix (using CW setting on WW element).

Paalap, 2023-06-01:
@marschr Thanks! This fixed it - now it si working fine.

(Only WhiteTenthMilliAmpere was ignored as I'm not using it and I miss a new additional settings class)

Makuna, 2023-06-05:
NOTE:  [Pull 710](https://github.com/Makuna/NeoPixelBus/pull/710) does a major refactor of features/NeoElements and fixes the ClearTo issue.  This is no longer compatible with your changes mentioned above.  But I will be re-creating your changes in the next few days and merge those in.

Makuna, 2023-06-05:
If you pick up the master branch, you will now have 
_NeoGrbcwxFeature_ - feature with color order of GRB Colder White, Warmer White, and an ignore channel x.
_RgbwwColor_ - color object that includes a WW (warmer white) and CW (colder white).  

**Note:** WW and CW are relative to each other rather than specific technical warm white and cold white.  So, if you have a neutral and cold, WW = neutral.  If you have a warm and neutral, then CW = neutral. 

If you have all three, WW, NW, and CW, then thats a whole other issue that needs to be created.

Paalap, 2023-06-06:
Thanks - is working very well fo me!

Will you also publish a release to arduino in the next time?

Makuna, 2023-06-27:
https://github.com/Makuna/NeoPixelBus/releases/tag/2.7.6

closed 2023-06-27 by Makuna

BLeibbrand, 2025-02-20:
I'm still having problems with my FW1906 strips.
The color order will be randomly picked by a manufacturer to get best strip layout for the copper lines.
So we should be able to easily set the colors after the above mentioned test to find the correct order.

[image]

Makuna, 2025-02-20:
@BLeibbrand Commenting on a closed issue is not a good process.  Its best to start a discussion or a new issue on the new concern/request and reference the closed issue.

A related issue that you may be running into is tracked by https://github.com/Makuna/NeoPixelBus/issues/804.