Makuna/NeoPixelBus #704: GPIO change at runtime does not invalidate the old GPIO in use.

sblantipodi, 2023-05-23:
**Describe the bug**
GPIO change at runtime does not invalidate the old GPIO in use and the old pins duplicates the signal sent to the new pin.

**To Reproduce**
Change the GPIO at runtime, as you can see now you have two GPIOs that runs the same commands.

```
 while (!ledsEsp32Rgbw->CanShow()) { yield(); }
    delete ledsEsp32Rgbw;
    ledsEsp32Rgbw = NULL;
    ledsEsp32Rgbw = new NeoPixelBus<NeoGrbwFeature, NeoSk6812Method>(dynamicLedNum, gpioInUse); // and recreate with new count
    if (ledsEsp32Rgbw == NULL) {
      Serial.println(F("OUT OF MEMORY"));
    } else {
      Serial.println();
      Serial.println(F("Initializing..."));
      while (Serial.available()) {
        Serial.read();
      }
      ledsEsp32Rgbw->Begin();
      ledsEsp32Rgbw->Show();
    }
```

**Expected behavior**
old GPIO should not be active after a GPIO change

**Development environment (please complete the following information):**
- Win 11 
- CLION with PlatformIO plugin
- I can reproduce the problem on every boards I have from ESP8266, to ESP32 to ESP32-S3.

Makuna, 2023-05-23:
This is specific to ESP32 isn't it?  ESP8266 doesn't have a muxer and really only works on one pin for each of the methods.

sblantipodi, 2023-05-23:
@Makuna yes it's specific to the ESP32 family, sorry for the wrong report of ESP8266.

Makuna, 2023-06-06:
RMT methods already disconnected it, so ESP32-S3/C3 should have already been working fine as that is the core method used.  
Fix applied for ESP32/ESP32C2

sblantipodi, 2023-06-07:
@Makuna I have retested it on my S3, and I agree that on S3 it works well already...

Makuna, 2023-06-27:
https://github.com/Makuna/NeoPixelBus/releases/tag/2.7.6

closed 2023-06-27 by Makuna

sblantipodi, 2023-06-27:
wonderful!