Makuna/NeoPixelBus #707: Add support for 8-bit color with WS2816

mcer12, 2023-05-27:
Default examples don't work because of this incompatibility. Adding a simple 8bit lookup table to RgbColor for high resolution chips (and thus allowing to use standard 8bit resolution) would make preliminary testing and adding support for these chips in projects much easier.

Or maybe support for 8bit NeoGrbFeature in NeoWs2816Method and do the conversion just before sending, whichever makes more sense.

Makuna, 2023-05-27:
Please follow the templates provided for issues and include the information it maps out.  You are missing so many details here.
Please provide exactly what definition you are using and how it's not working.  Be complete and concise. 

RgbColor can be used directly when calling strip.SetPixelColor().  strip.GetPixelColor returns a 16 bits per element Rgb48Color thus will not auto-convert to an 8-bit due to a loss of information from 16 down to 8 bits.  It is a bad idea to lose information for a sketch writer without them knowing so it will never do this.  You have to explicitly convert it.

Makuna, 2023-06-07:
...waiting on specifics...

Makuna, 2023-07-09:
If I don't hear a response, I will be closing this issue.

To get the pixel color, you simply have to call GetPixelColor like this...
```
    RgbColor color = RgbColor(strip.GetPixelColor( 1 ));
```

closed 2023-07-17 by Makuna