Makuna/NeoPixelBus #714: Timing changes

scarr999, 2023-06-12:
Make slight changes to bit timings so the library will work with these popular LED string https://www.ebay.co.uk/itm/373588834942?var=642594617777

I have tried many libraries but all fail

I do have logic analyzer screen shots attached and here are the timings 

T1H 320ns
T1L  1.28us
T0H  1.12us
T0L  480ns

it does have a 9.12us gap between LED packets, I can provide the actual capture file and you an download the free Saleae software to view it.

happy to pay to have a one off library or contribute if it becomes part of the main library

[image]

Makuna, 2023-06-12:
What Arduino/uC are you using?

Also need the reset time, the time between the last pixel of current frame and the first pixel of the next frame.

You did try the `NeoRgbwColor` and `NeoSk6812Method` when you declared your NeoPixelBus?

Also try the `NeoApa106Method` as it is close to the same timing that you included above.

Interpixel gaps can't be supported by all methods at all and only one method core supports it today, Esp8266 Dma methods.  But in a previous investigation on a similar pulse form we found that the interpixel gap wasn't a requirement, it was a artifact of the crappy controller they included.

scarr999, 2023-06-13:
Hi Michael,

Thanks for the reply, I did try the code below both with NeoRgbwColor and
NeoSk6812Method, NeoSk6812Method compiled but didn't work and NeoRgbwColor
didn't compile saying "Compilation error: 'NeoRgbwColor' was not declared
in this scope"

// NeoPixelTest
// This example will cycle between showing four pixels as Red, Green, Blue,
White
// and then showing those pixels as Black.

#include <NeoPixelBus.h>

const uint16_t PixelCount = 4; // this example assumes 4 pixels, making it
smaller will cause a failure
const uint8_t PixelPin = 3;  // make sure to set this to the correct pin,
ignored for Esp8266

#define colorSaturation 128

NeoPixelBus<NeoGrbFeature, NeoRgbwColor> strip(PixelCount, PixelPin);

//NeoPixelBus<NeoRgbwFeature, NeoSk6812Method> strip(PixelCount, PixelPin);

RgbColor red(colorSaturation, 0, 0);
RgbColor green(0, colorSaturation, 0);
RgbColor blue(0, 0, colorSaturation);
RgbColor white(colorSaturation);
RgbColor black(0);

HslColor hslRed(red);
HslColor hslGreen(green);
HslColor hslBlue(blue);
HslColor hslWhite(white);
HslColor hslBlack(black);

void setup()
{
    Serial.begin(115200);

    Serial.println();
    Serial.println("Initializing...");
    Serial.flush();

    // this resets all the neopixels to an off state
    strip.Begin();
    strip.Show();

    Serial.println();
    Serial.println("Running...");
}

void loop()
{
    delay(5000);

    Serial.println("Colors R, G, B, W...");

    // set the colors,
    // if they don't match in order, you need to use NeoGrbFeature feature
    strip.SetPixelColor(0, red);
    strip.SetPixelColor(1, green);
    strip.SetPixelColor(2, blue);
    strip.SetPixelColor(3, white);
    // the following line demonstrates rgbw color support
    // if the NeoPixels are rgbw types the following line will compile
    // if the NeoPixels are anything else, the following line will give an
error
    // strip.SetPixelColor(3, RgbwColor(colorSaturation));
    strip.Show();

    delay(5000);

    Serial.println("Off ...");

    // turn off the pixels
    strip.SetPixelColor(0, black);
    strip.SetPixelColor(1, black);
    strip.SetPixelColor(2, black);
    strip.SetPixelColor(3, black);
    strip.Show();

    delay(5000);

    Serial.println("HSL Colors R, G, B, W...");

    // set the colors,
    // if they don't match in order, you may need to use NeoGrbFeature
feature
    strip.SetPixelColor(0, hslRed);
    strip.SetPixelColor(1, hslGreen);
    strip.SetPixelColor(2, hslBlue);
    strip.SetPixelColor(3, hslWhite);
    strip.Show();

    delay(5000);

    Serial.println("Off again...");

    // turn off the pixels
    strip.SetPixelColor(0, hslBlack);
    strip.SetPixelColor(1, hslBlack);
    strip.SetPixelColor(2, hslBlack);
    strip.SetPixelColor(3, hslBlack);
    strip.Show();

}

I have attached a Saleae capture from the working ebay controller that
should allow you to see everything including all timings. The Saleae
software is free to download and use Software
<https://logic2api.saleae.com/download?os=windows>
This capture is a pulse brightness cycling colours.

Thanks again for your reply and help.

scarr999, 2023-06-14:
> I meant NeoRgbwFeature together with NeoSk6812Method. Your commented out
> line suggest this is what you were using.
>
> The attachment wasn't made. I don't believe you can include attachments;
> you can imbed images, or you link to external files that you share using a
> service.
>
> JFYI: This part of the Wiki is helpful in the details
>
> You didn't mention if you had tried the NeoApa106Method method.
>
> NeoPixelBus<NeoRgbwFeature, NeoApa106Method> strip(PixelCount, PixelPin);

Hi Michael,

I have tried

*NeoPixelBus<NeoRgbwFeature, NeoApa106Method> strip(PixelCount, PixelPin);*

And still nothing, be aware the code I sent and am using is just what I
downloaded from advice on a forum, I am just stabbing in the dark a little
bit as this project is for a festival and I am running out of time! The
time I would normally take to fully understand it all :-( I will look
at "*NeoRgbwFeature together
with **NeoSk6812Method*" today.

When I looked at the SK6812 datasheet and the data capture, to me it looks
like the bit timings on these ebay LED'S are out of spec, however I might
be wrong and I'm sure you looking at the data stream would quickly be able
to say YES/NO, I have added a link to the capture below

Capture file
<https://drive.google.com/file/d/1ymTzq5M-Cv76v-0opH1eyaHxXhQmpNj4/view?usp=sharing>

Thanks again.

Makuna, 2023-06-14:
What Arduino are you using? AVR?  ESP32?

Makuna, 2023-06-14:
The capture shows a different timing and variable.  From what I see:

0 - 320us-480us high with total pulse size of between 1440us and 1600us
1 - 960us-1120us high with a total pulse size of between 1600us and 1760us
reset time unknown, but they are waiting 14ms between or 14000000us

usually these are output as ratio of total pulse, with a consistent pulse pw
0 = 0.25 * pw and 
1 = 0.75 * pw

using the highest of 0 and lowest of 1 for pw
1600 * 0.25 = 400us (seems fine)
1600 * 0.75 = 1200us (a little long)

But each platform has unique timing restrictions so I can't proceed without knowing the micro controller (uC).

scarr999, 2023-06-14:
Hi, Arduino nano

scarr999, 2023-06-15:
Hi,

I have added another file to the cloud that will hopefully give you the
reset timing, to me it appears to be 1.2ms Reset delay
<https://drive.google.com/file/d/1cRtK_K6qX878Cnxtzleh_6ZyUpX_vKRO/view?usp=sharing>

Is NeoRgbwColor a feature or method? I get compilation errors when trying
it. also the original packet I am sure is RGB,  if RGBW has more data for
white then I am sure this will not work.

I did try "NeoPixelBus<NeoRgbFeature, NeoApa106Method> strip(PixelCount,
PixelPin);" but it didn't work either.

I agree with the interpixel gaps as these should not be required.

Steve

scarr999, 2023-06-15:
> The issue is that you are using a nano. The Sk6812 should have gave you
> timing close to what you need. But the AVRs use hand written assembly code
> to time the pulses, and there is only a standard pulse ratio and timing on
> that platform. So, it will require another instance of hand written
> assembly to truly support sk6812 spec'ed timing.

Hi,

I'm happy to pay for this if that helps?

Makuna, 2023-06-15:
$2000 an hour?  8-)

This branch contains a new Apa106 timing (for 16Mhz AVRs only).  Your Nano is running at 16Mhz correct?
https://github.com/Makuna/NeoPixelBus/tree/AVRSk6812
Check the readme and you can clone directly into your libraries folder replacing your NeoPixelBus.

Use the method `NeoApa106Method`.

Makuna, 2023-06-19:
So, as it sits right now, you see nothing using the Nano?  
Do you know the logic level of the original, what voltage the pulse is?  The Nano will be 5v (unless yours is a 3.3v one but I thought those ran at a lower speed).  
I will look at adding the inter-pixel delay next.  Normally this isn't a requirement but an artifact of a uC in the controller; but as it sits now if you don't see anything, it sounds like it is.

scarr999, 2023-06-19:
Hi, Yes, 5v as seen on the saleae logic traces and yes its a 5v nano.

I see nothing not even a flicker.

I actually thought the same as you, so in an attempt to prove this, changed
the code to output just one led's data hoping this would show up because
there was no interpixel delay needed.

Still nothing! Tomorrow I will check data out with the single LED packet,
se if its getting past the first LED meaning that the pulse widths are
still not right.

Thanks

Makuna, 2023-06-21:
Ok, where does this leave us?
The chip is repeating the signal to the next LED, but is not showing anything.
Could you capture the output using the latest for a single LED?  (make sure to include the amount of time between sends).

Makuna, 2023-06-21:
Pick up the latest changes to the working branch.
Use the `NeoAvr600KbpsIpsMethod` method.  This adds the interpixel delay so it should now mimic the original near exactly.

scarr999, 2023-06-22:
Hi,

Still not working, I have made a new folder with todays data in it, it's called 22_06_2023

I installed the library and complied, I then did a Saleae capture (MyCode_Capture_22_06_2023)  I can see the interpixel delays but nothing on the LED’S. 
 
I can see the same data at LED1 as I can at LED4, if I understand the SK6812 protocol the data should reduce by one 24bit packet after each LED as the LED removes it’s data and passes the rest on. but maybe I have misunderstood this? 

I also have added excel spreadsheets that may make it easier to look at timings if you’re not using the Saleae software that is to view the actual capture.

**Saleae capture files:**

- MyCode_Capture_22_06_2023  
                  My code outputting 4 x RED (0x48) packets 

- Original_Pulsing_RED 

                  Capture from the original controller output pulsing LED’s as shown on the previous video. You have to hunt for a 
                  RED (0x48)

- Original_pulsing_RED_Two_packets

	Two full LED string packets from the original controller captured at the end of the string.

**Excel spreadsheets:**

- My_Code_Capture_22_06_2023 

        The “MyCode_Capture_22_06_2023” capture data exported and formatted for ease.

- Original_pulsing_RED_Two_packets

        The “Original_pulsing_RED_Two_packets” capture data exported and formatted for ease.

- Side By Side Comparison

        One LED's data from both original and mycode including interpixel delay at the end, each has a value of 0x48 and shows 
        the difference in timings.

Thanks a million for not giving up, this is what I love about coding, figuring stuff out, with a lot of help ;-)

Makuna, 2023-06-22:
Tuned ASM to get tighter signal matching.  
Byte boundary stretching of the pulse is within tolerance.
1 bit high lengthened a little to better sit in the middle of tolerance.

Pickup the changes and give it a try, along with other things previously mentioned.

scarr999, 2023-06-23:
Hi,

**One thing I noticed from your capture that I do not see when I capture using my library is that the pulse width after a byte (every eighth pulse) is larger; being over 2us rather than 1.6 us.**

Not quite sure what you mean have you got a picture as I can't see it.

If we are saying that when sending 10 LED packets to a 10 LED string at the end of a string you should see no data?, This is where there is an issue as I it appears that the data is not decreasing even on the original controller. I have made a new folder on the cloud, (23_06_2023) in it are three captures, 

one without LEDS attached and sampled at the controller
one with LEDS attached and sampled at the controller
one with LEDS attached and sampled at the end of the string

Apart from when the LEDS aren't attached they look the same at the start and end of the string., 200 LED packets, I assume the inverted data is due to no pulldown when the LEDS are attached.

Steve

Makuna, 2023-06-23:
I wish we knew the actual model number of the LEDs/Chips.  The sellers site states SK6812 but they are not following the protocol for those.  This is something else.

So, you have to double the output?  What actual NPB pixel index is the one used for the output color?  Is the first pixel color set by calling SetPixelColor(0, color) or by SetPixelColor(193, color)?  
I have seen Two Wire (DotStar) LEDs that need extra "bytes" to push/latch the signal through based on the string count and my library does that without using extra memory for a buffer.  Maybe this is a similar case.  If so I can adjust the code.  Let me know.

You could link your project site within the [Wiki here](https://github.com/Makuna/NeoPixelBus/wiki/Project-References)
You could write a "topic tutorial" for the [Wiki Here](https://github.com/Makuna/NeoPixelBus/wiki/Advanced-Topics-Tutorials)
You could post a discussion linking a blog on Github.

scarr999, 2023-06-26:
You master library works! using so it appears not to be a timing issue but the fact these LED's are addressed!
    NeoPixelBus<NeoGrbFeature, NeoSk6812Method> strip(PixelCount, PixelPin);

I originally cut 4 LED's from the end of the string (199,198,197,196), and after trying to get these to work I thought I might have blow them so cut a further 4 LED's (195,194,193,192) from the string and these are currently being used.

Having just 4 LED's I assumed! that I could set the pixel count to 4 and control these 4 LED's. (1,2,3,4)

However to get these 4 attached LED's to work I have to set the Pixelcount to 200 and then use the following and they light up correctly
      strip.SetPixelColor(192, red);
      strip.SetPixelColor(193, green);
      strip.SetPixelColor(194, blue);
      strip.SetPixelColor(195, red);

I can also set the pixel count to 400 and it still works fine.

Steve

scarr999, 2023-07-07:
Hi,

What's the best way of monitoring the status of you looking at these LED strings when they arrive?

Makuna, 2023-07-07:
Tracking states by Sept 15th.  I suspect they will get here way before that.
I pinned this issue (shows in a box at the top of issues); do you see that?

scarr999, 2023-07-08:
Thanks, dam that's a long time! Maybe as you say quicker, was hoping to get
it sorted for a festival custume but hey ho,

I made a donation yesterday and I'd like to thank you again, best support
I've ever had 👍👍

scarr999, 2023-07-14:
FYI, I have 100 confirmed they are allocated a number. When lighting LED 1 the same LED lights regardless of what end of the string I connect the wires to.

Makuna, 2023-07-19:
@scarr999 They finally arrived.  Now I need to schedule some time to investigate.

With your last statement, are you stating that you set the signal input to the other side (what may be called the output from the shipped wiring) and it still worked?!

scarr999, 2023-07-20:
@Makuna 
Yes, I removed the OEM controller so I had just a length of LEDs no matter what end I connect to, if I turn LED 1 RED the same LED lights up. this indicates to me they are pre-allocated an address.

scarr999, 2023-07-20:
> Thats a new twist, either the data wire is parallel to the LEDs or they
> can support input/output swapping.

Does this not go with what we found? each led does not take its data from
the stream, it simply sees it? So at either end you see the full 200 leds
data.

That way there is no in/out just a data line.

Makuna, 2023-07-25:
Confirmed that they are not only individually addressed, but that connecting it to the other side doesn't change the order.  This means individually, the look for their own nth data item in a parallel stream of data.  This is not like any other RGBIC out there that I know of.  I have yet to find a way to reset the ID and so far, the only Hints provided to me from sellers is that each led/chip is unique.  This may mean that they may not be reprogrammable.

Something I noticed is that the voltage on the far side of a 200 LED strip drops to near 3v from 5v; but they still seem to work.

scarr999, 2023-07-25:
Hi,

Good to know I'm not loosing the plot :-)

I realise it might be impossible to figure out how to reprogram these, I'm not sure how I'd start tbh. I have new ones arriving, slightly different make but very similar, I will do the initial analysis based on what I have learnt and report back here .

Makuna, 2023-11-20:
So, summarizing...
These LEDs are individually identified, so the nth will always be the nth no matter if the strip is cut or the signal is fed from the other side.
They are functional using  NeoWs2812xMethod with the knowledge of the above.
I have found no way to software reset the id of each LED.

Not sure there is more I can do other than document this well in the Wiki.