Makuna/NeoPixelBus #717: trademark misuse

Universal-Solder, 2023-06-13:
"APA106, SK6812, WS2811, WS2812 and WS2813 that are commonly refered to as NeoPixels"

NO!

NeoPixel is a trademarked term by Adafruit Industries and has nothing to do with the parts mentioned. It is Adafruit's advertising that makes people believe these parts were "NeoPixel", which is just as stupid as calling every tissue a Kleenex.

Makuna, 2023-06-13:
Adafruit knows about this library since close to when it was created and has used code from it for their library to support ESP8266 in the early days of that platform.  
They have not requested any name changes or edits to enforce their trademark.

People still refer to facial tissues as Kleenex, just like people still refer to these chips generally as NeoPixels.

There is no universally used industry term for them that is succinct and widely known as it.  This is why that specific line of text is used.  

Why did you create this issue?

Universal-Solder, 2023-06-13:
Thank you for the explanation.
I have been in trouble with Adafruit (years ago) over NeoPixel. They had my
products removed from eBay and Amazon because I used the term NeoPixel in
combination with WS2812B products. Even "NeoPixel compatible" was not good
enough for them.
I must apologize, I shouldn't have created this issue. It is just that I
still see a red rag when I see that someone supports Adafruit by using
their brand name instead of calling the product by its actual name.

Makuna, 2023-06-13:
Thank you for sharing.  I suspect it might be that you were selling a product while I am not.  

Over time, I have been moving to more specific names like WS2812x/APA106/etc. But there needs to be an open industry standard name beyond NeoPixel, this I agree.  

I often see ICLED and IALED but the acronyms are confusing because they are not used consistently.  I have seen ICLED used with a strip that while the whole strip was color controlled, the individual LEDs were not.

ICLED = individually controlled LED
IALED = individually addressed LED

Also, I have seen RGB LED, but this doesn't convey controllability versus RAW RGB LEDs.  Confusing buyers worse.

There is also the issue with DotStar (again Adafruit). I have seen Two Wire and Three Wire be used (do you include the GND pin?).  I have also seen Clocked Signal LED used; which is more descriptive to its uniqueness.  I believe FastLED uses the term Clocked in this case.

I will leave this open for tracking any new information on the topic.  I was going to move it over to a discussion; but you have valid points if Adafruit wanted to ask me to change.