Makuna/NeoPixelBus #729: error: 'NOT_A_PIN' was not declared in this scope

kurogedelic, 2023-08-09:
Thanks for a wonderful library !
I tried "NeoPixelTest.ino" build for SAMD21 architectures e.g. Seeed XIAO M0, Arduino M0.
there's compile error.

```
In file included from /private/var/folders/gs/fy565xcd6w5bv4vq3hcrzpbw0000gn/T/.arduinoIDE-unsaved202379-5579-tlr700.2kho/NeoPixelTest/NeoPixelTest.ino:14:0:
/Users/[user]/Documents/Arduino/libraries/NeoPixelBus_by_Makuna/src/NeoPixelBus.h:73:118: error: 'NOT_A_PIN' was not declared in this scope
     NeoPixelBus(uint16_t countPixels, uint8_t pinClock, uint8_t pinData, uint8_t pinLatch, uint8_t pinOutputEnable = NOT_A_PIN) :
                                                                                                                      ^~~~~~~~~
/Users/[user]/Documents/Arduino/libraries/NeoPixelBus_by_Makuna/src/NeoPixelBus.h:73:118: note: suggested alternative: 'NOT_A_PORT'
     NeoPixelBus(uint16_t countPixels, uint8_t pinClock, uint8_t pinData, uint8_t pinLatch, uint8_t pinOutputEnable = NOT_A_PIN) :
                                                                                                                      ^~~~~~~~~
                                                                                                                      NOT_A_PORT

exit status 1

Compilation error: exit status 1
```

thanks

Makuna, 2023-08-09:
This is a bug for the board package owner.  I searched, and for some reason they use PIN_NOT_A_PIN instead of the standard NOT_A_PIN.  I would suggest raising the issue with them.

In the mean time, before you include NeoPixelBus, just add this...
```
#define NOT_A_PIN PIN_NOT_A_PIN
```
and it should work.

kurogedelic, 2023-08-12:
I added in my .ino "#define NOT_A_PIN PIN_NOT_A_PIN" but I get a similar error. (on any line)
Replacing it with "NOT_A_PORT" by IDE suggestion also fails.

I will report it to the board package project.
thanks.

Makuna, 2023-08-12:
I added a workaround in this pull.  https://github.com/Makuna/NeoPixelBus/pull/730

Makuna, 2024-03-19:
Also added where there was no alternative to map it to -1.

closed 2024-03-19 by Makuna