Makuna/NeoPixelBus #747: Add support for TM1934

bogdanr, 2023-12-01:
Hello

I have a TM1934 LED strip. The online listing specifies that this is similar to WS2815.
[image]

I tried with some of the currently supported variants but it doesn't work properly. For example TM1814 doesn't really hold a steady color with this chip and using WS2813 allows for very little brightness.

Makuna, 2023-12-04:
@bogdanr Let me know if using the TM1914 works for you and then I can alias the TM1934 to it.

bogdanr, 2023-12-05:
I am using esp8266, the following config:
```
...
esp8266:
  board: d1_mini_pro

light:
  - platform: neopixelbus
    variant: TM1914
    id: $id
    pin: D4
    num_leds: 125
    type: GRB
    color_correct: [30%, 30%, 30%]
    name: "$devicename Light"
#    restore_mode: ALWAYS_ON
    on_turn_on:
      then:
        - light.control:
            id: $id
            effect: "Rainbow Spinner"
    effects:
      - addressable_rainbow:
          name: "Rainbow Spinner"
          speed: 8
          width: 256
      - addressable_rainbow:
          name: "Rainbow Fader"
          speed: 3
          width: 256
      - pulse:
      - strobe:
      - addressable_flicker:
      - addressable_color_wipe:
      - addressable_scan:
      - addressable_twinkle:
```
With TM1914 it doesn't work properly in the sense that LED brightness is 100% unless it's completely OFF. If I use effects like color_wipe it changes the LEDs but the ones that don't show some color are 100% white. Selecting no effect with 1% brighness it's white 100% and only with 0% brighness it's off.

Setting the variant WS2811, the effects work correctly (LEDs are actually turning off and not going to 100% when they should be off) but the maximum brightness is at most 10% and it's not adjustable.

bogdanr, 2023-12-05:
Actually the problem was with `color_correct: [30%, 30%, 30%]`. 
With the following config it works just fine so if you can set the alias to ws2812x that would be great.
```
    variant: ws2812x
    pin: D4
    num_leds: 125
    type: RBG
```

closed 2023-12-05 by bogdanr