Makuna/NeoPixelBus #764: Support for WS2805

Osmio0075, 2024-02-07:
**Describe the solution you'd like**
NeoPixelBus should support communicating with the WS2805 and manage the five channels it has.

**Additional context**
The WS2805 is a relatively new chip, which is starting to show up in some commercial products. It would be useful to the community being able to control these new LED.

I've got a datasheet from [here](https://datasheet.lcsc.com/lcsc/2306301035_Worldsemi-WS2805_C5446697.pdf)

And I found out about this new chip in this [Aliexpress page](https://aliexpress.com/item/1005006420002446.html).

Makuna, 2024-02-08:
Added NeoRgbwwFeature and common variant orders of the RGB elements.

https://github.com/Makuna/NeoPixelBus/pull/765

Makuna, 2024-03-15:
May require a slower data rate, spec states 600Kbps.  The APA106 could be used as the timing is the same but the reset time is not long enough.

Makuna, 2024-03-17:
Add slower rate methods for Ws2805 if the normal Ws2812x doesn't work.

Makuna, 2024-03-19:
https://github.com/Makuna/NeoPixelBus/releases/tag/2.7.9

closed 2024-03-19 by Makuna