Makuna/NeoPixelBus #771: "Loop Sends" for repeating output stream on low memory builds.

sparkplug23, 2024-02-17:
**Is your feature request related to a problem? Please describe.**
As discussed on Discord, sending the data stream multiple times to replicate the output. This would be particularly useful in low memory applications.

**Describe the solution you'd like**
An optional parameter to enable repeating pixel buffer multiple times. Perhaps with optional arg on begin, or simply a function that should be called during runtime (similar to usage settings). Runtime config would be easier than setting it when the class is defined.  Since this is an optional feature, it may just be easier (memory efficient) to keep it as a build/template option. 

In my personal use of this library, when I need to maintain FPS and have large data outputs (1000+ pixels) I have introduced a "decimate" feature. For example, if I had 1000 LEDs physically connected but the effect running only calculated the first 100 internal pixels then repeated this in sequence, instead of regenerating the internal colours from the effect/palette code, I would replicate the first 100 pixels 10 times (with SetPixelColour). This case would be decimate of 10% (ie only 10% of the output needs to be calculated). You could perhaps introduce this type of feature natively in NeoPixelBus as my repeated output is essentially this feature.

Makuna, 2024-04-16:
BTW, what platform are you using?  I might have an alpha branch up shortly for AVR that supports this.  ESP32 is next on the list.

sparkplug23, 2024-04-16:
I am mostly using esp32 these days as I work on my code, though will eventually compiling again on esp8266. I have added my own colour type into your library so it takes me some time to merge with your latest releases. 

I will be able to test on the attiny85, this is what sparked this original conversation.