Makuna/NeoPixelBus #776: NeoPixelBus 2.7.8 example NeoPixelTest.ino will not compile for an ESP32-C3 using Arduino IDE 1.8.19

HenkHoldijk, 2024-03-07:
In file included from ...\NeoPixelBus_by_Makuna\src/internal/NeoMethods.h:56,
                 from ...\NeoPixelBus_by_Makuna\src/NeoPixelBus.h:43,
                 from ...\NeoPixelBus_by_Makuna\examples\NeoPixelTest\NeoPixelTest.ino:14:
...\NeoPixelBus_by_Makuna\src/internal/methods/NeoEsp32RmtMethod.h: In destructor 'NeoEsp32RmtMethodBase<T_SPEED, T_CHANNEL>::~NeoEsp32RmtMethodBase()':
...\NeoPixelBus_by_Makuna\src/internal/methods/NeoEsp32RmtMethod.h:559:9: error: there are no arguments to 'gpio_matrix_out' that depend on a template parameter, so a declaration of 'gpio_matrix_out' must be available [-fpermissive]
  559 |         gpio_matrix_out(_pin, 0x100, false, false);
      |         ^~~~~~~~~~~~~~~

...\NeoPixelBus_by_Makuna\src/internal/methods/NeoEsp32RmtMethod.h:559:9: note: (if you use '-fpermissive', G++ will accept your code, but allowing the use of an undeclared name is deprecated)
...\NeoPixelBus_by_Makuna\src/internal/methods/NeoEsp32RmtMethod.h: In instantiation of 'NeoEsp32RmtMethodBase<T_SPEED, T_CHANNEL>::~NeoEsp32RmtMethodBase() [with T_SPEED = NeoEsp32RmtSpeedWs2812x; T_CHANNEL = NeoEsp32RmtChannel1]':
...\NeoPixelBus_by_Makuna\src/NeoPixelBus.h:55:9:   required from 'NeoPixelBus<T_COLOR_FEATURE, T_METHOD>::NeoPixelBus(uint16_t, uint8_t) [with T_COLOR_FEATURE = NeoGrbFeature; T_METHOD = NeoEsp32RmtMethodBase<NeoEsp32RmtSpeedWs2812x, NeoEsp32RmtChannel1>; uint16_t = short unsigned int; uint8_t = unsigned char]'
...\NeoPixelBus_by_Makuna\examples\NeoPixelTest\NeoPixelTest.ino:22:72:   required from here
...\NeoPixelBus_by_Makuna\src/internal/methods/NeoEsp32RmtMethod.h:559:24: error: 'gpio_matrix_out' was not declared in this scope; did you mean 'gpio_iomux_out'?
  559 |         gpio_matrix_out(_pin, 0x100, false, false);
      |         ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |         gpio_iomux_out

Could be related to : https://github.com/espressif/esp-idf/issues/11737

Makuna, 2024-03-07:
What version of the ESP32 board support are you using?  With the latest released version, I cannot replicate the error above.  You can check the Board Manager for the version.

What board are you selecting?  Check the Tools Board menu.

Thanks for the link, I am applying that correction now.

Makuna, 2024-03-07:
https://github.com/Makuna/NeoPixelBus/pull/777

HenkHoldijk, 2024-03-07:
Hi,

I'm using 3.0.0-alpha3 (as I need the support for the ESP32-C3). In the Board manager select the ESP32C3 Dev Module.

Cheers, Henk

Makuna, 2024-03-07:
I am in the middle of some other work and can't switch to v3.0.0-alpha3 at the moment.  I wish they would make a release with it marked as beta and not default so that it's easy to switch to it and back.

If you could pick up my master and give it a try and let me know if there is anything else, that would help.

HenkHoldijk, 2024-03-08:
Will give your master a try this evening.

HenkHoldijk, 2024-03-08:
Sorry, but only a slightly changed error.

In file included from ...\libraries\NeoPixelBus_by_Makuna\src/internal/NeoMethods.h:56,
                 from ...\libraries\NeoPixelBus_by_Makuna\src/NeoPixelBus.h:43,
                 from ...\libraries\NeoPixelBus_by_Makuna\examples\NeoPixelTest\NeoPixelTest.ino:14:
...\libraries\NeoPixelBus_by_Makuna\src/internal/methods/NeoEsp32RmtMethod.h: In destructor 'NeoEsp32RmtMethodBase<T_SPEED, T_CHANNEL>::~NeoEsp32RmtMethodBase()':
...\libraries\NeoPixelBus_by_Makuna\src/internal/methods/NeoEsp32RmtMethod.h:559:9: error: there are no arguments to 'gpio_matrix_out' that depend on a template parameter, so a declaration of 'gpio_matrix_out' must be available [-fpermissive]
  559 |         gpio_matrix_out(_pin, SIG_GPIO_OUT_IDX, false, false);
      |         ^~~~~~~~~~~~~~~
...\libraries\NeoPixelBus_by_Makuna\src/internal/methods/NeoEsp32RmtMethod.h:559:9: note: (if you use '-fpermissive', G++ will accept your code, but allowing the use of an undeclared name is deprecated)
...\libraries\NeoPixelBus_by_Makuna\src/internal/methods/NeoEsp32RmtMethod.h: In instantiation of 'NeoEsp32RmtMethodBase<T_SPEED, T_CHANNEL>::~NeoEsp32RmtMethodBase() [with T_SPEED = NeoEsp32RmtSpeedWs2812x; T_CHANNEL = NeoEsp32RmtChannel1]':
...\libraries\NeoPixelBus_by_Makuna\src/NeoPixelBus.h:55:9:   required from 'NeoPixelBus<T_COLOR_FEATURE, T_METHOD>::NeoPixelBus(uint16_t, uint8_t) [with T_COLOR_FEATURE = NeoGrbFeature; T_METHOD = NeoEsp32RmtMethodBase<NeoEsp32RmtSpeedWs2812x, NeoEsp32RmtChannel1>; uint16_t = short unsigned int; uint8_t = unsigned char]'
...\libraries\NeoPixelBus_by_Makuna\examples\NeoPixelTest\NeoPixelTest.ino:22:72:   required from here
...\libraries\NeoPixelBus_by_Makuna\src/internal/methods/NeoEsp32RmtMethod.h:559:24: error: 'gpio_matrix_out' was not declared in this scope; did you mean 'gpio_iomux_out'?
  559 |         gpio_matrix_out(_pin, SIG_GPIO_OUT_IDX, false, false);
      |         ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |         gpio_iomux_out

Cheers, Henk

Makuna, 2024-03-08:
thanks, I will take a closer look in a week or so when I clear some other things off my table.

Makuna, 2024-03-22:
They have made major changes to both RMT and I2S interfaces of which this library relies on both.  It will require significant rewriting and refactoring to remove deprecation warnings.

Makuna, 2024-03-22:
https://github.com/Makuna/NeoPixelBus/pull/789

oehoe, 2024-04-23:
Hello, I have a similar issue. 

Compiling the following code with both esp32 Board c3.0.0-rc1 and 2.0.11 gives me a compilation error.
Neopixelbus 2.7.9 on Mac OS Sonoma on M2 MacBook Air. Selected board is XIAO_ESP32C3. I have read all about the issues with these boards but RMT method should be working?

```
#include <NeoPixelBus.h>

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

```
****/Documents/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/methods/Esp32_i2s.c:50:10: fatal error: soc/sens_reg.h: No such file or directory
   50 | #include "soc/sens_reg.h"
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: exit status 1
```

Makuna, 2024-04-23:
> Hello, I have a similar issue.
> 

Did you pick up the master branch and did it allow you compile?  This issue is still marked "pending", meaning its present in "master" but not released yet.  The next release will include it, which is soon.

oehoe, 2024-04-24:
Thanks for your reply. I did download the master branch. I still get the compile error.

Makuna, 2024-04-24:
> Thanks for your reply. I did download the master branch. I still get the compile error.

ESP32C3 Dev Kit compiles fine.
XIAO_ESP32C3 does reproduce your issues.  I will look into it.

Makuna, 2024-04-28:
https://github.com/Makuna/NeoPixelBus/releases/tag/2.8.0

closed 2024-04-28 by Makuna