Makuna/NeoPixelBus #800: False positive error report

BlueAndi, 2024-04-24:
The "NeoEsp32RmtMethod" is used to update the strip.
After a Show(), the CanShow() is called periodically to determine that the strip was updated.

CanShow() calls IsReadyToUpdate() method:
https://github.com/Makuna/NeoPixelBus/blob/e40289f89c1fc869ca68b1622efbb49866f8b01e/src/internal/methods/NeoEsp32RmtMethod.h#L598-L601

The rmt_wait_tx_done() is called with a timeout of 0, which causes an error report in case the transfer is not finished yet.
See
[image]

This pollute the log and causes the slow down the system.

Could we skip using the ESP_ERROR_CHECK_WITHOUT_ABORT() in especially this case?

---
NeoPixelBus: v2.7.9
Platform: espressif32 v6.5.0
Framework: arduino
Board: esp32-S3 DevKitC-1

Makuna, 2024-04-24:
The timeout shouldn't be an error, or they should provide a routine you can call to check/poll the state.  

But your suggestion sounds like a good work around.

Makuna, 2024-04-25:
https://github.com/Makuna/NeoPixelBus/tree/Esp32RmtSilentTimeout contains the changes to silence timeouts.

Give it try, if it works I will merge into master before the next release.

BlueAndi, 2024-04-25:
Thanks, will try and come back with feedback.

BlueAndi, 2024-04-25:
Successful tested, you can integrate the bugfix. Thanks for the fast and good support!!

Makuna, 2024-04-28:
https://github.com/Makuna/NeoPixelBus/releases/tag/2.8.0

closed 2024-04-28 by Makuna