Makuna/NeoPixelBus #813: Use Native type for loops that have large interations to save code size

Makuna, 2024-07-13:
**Is your feature request related to a problem? Please describe.**
WLED developers have found that using a native size type to the uC (ESP32 would be 32 bit) on loops can reduce code size.  

**Describe the solution you'd like**
Define a NeoPixelBus wide type like `uintLoop_t` and `intLoop_t` (alternate of `uintNative_t` or others?) that then get used throughout NeoPixelBus.  They get defined based on the microcontrollers platforms size.  (using int and unsigned int?)

**Additional context**
This seems to be a bigger issue (using smaller size than the native uC size) for larger chips as they then need to inject code on every increment or test to convert a native register to the size the code defined.

blazoncek, 2024-07-13:
Perhaps just `size_t` would do when you only need positive numbers.

blazoncek, 2024-07-13:
https://en.cppreference.com/w/cpp/types/integer

softhack007, 2024-09-20:
You could use the "fast" types like `uint_fast16_t` or `int_fast32_t` (see list in the link posted by @blazoncek).

Some people consider these make code look "nerdy" however it's the most portable solution. Like `uint_fast8_t` is "the fastest (native) unsigned that can hold at least 8 bits".