Makuna/NeoPixelBus #816: Doesn't compile on Arduino IDE V2.3.2 for an RP2040.

FlightDataRecorder, 2024-07-20:
Installed NeoPixelBus V2.8.0, tried to compile NeoPixel_RP2040_PioX4.ino and it blows up on these lines...

NeoPixelBus<NeoBgrFeature, NeoRp2040Pio1X4Ws2811Method> strip1(120, 15); // note: older WS2811 and longer strip
NeoPixelBus<NeoGrbFeature, NeoRp2040Pio1X4Ws2812xMethod> strip2(100, 2); // note: modern WS2812 with letter like WS2812b
NeoPixelBus<NeoGrbFeature, NeoRp2040Pio1X4Ws2812xInvertedMethod> strip3(100, 4); // note: inverted
NeoPixelBus<NeoGrbwFeature, NeoRp2040Pio1X4Sk6812Method> strip4(50, 16); // note: RGBW and Sk6812 and smaller strip

If I comment the each one out it just crashes on the next one complaining that NeoRp2040Pio1X4Ws2811Method was not declared in this scope.

Any help would be appreciated; I went and bought ~$150 worth of SK6812 Strips and Power Supplies based on this page https://www.arduino.cc/reference/en/libraries/neopixelbus-by-makuna/ saying "This library is compatible with ALL architectures."

Thank you.

Makuna, 2024-07-20:
The example was not updated after changes for RP2040 name convention was moved to the modern pattern.  Try these (copy and paste into the sketch to replace the definitions).
```
NeoPixelBus<NeoBgrFeature, Rp2040x4Pio1Ws2811Method> strip1(120, 15); // note: older WS2811 and longer strip
NeoPixelBus<NeoGrbFeature, Rp2040x4Pio1Ws2812xMethod> strip2(100, 2); // note: modern WS2812 with letter like WS2812b
NeoPixelBus<NeoGrbFeature, Rp2040x4Pio1Ws2812xInvertedMethod> strip3(100, 4); // note: inverted
NeoPixelBus<NeoGrbwFeature, Rp2040x4Pio1Sk6812Method> strip4(50, 16); // note: RGBW and Sk6812 and smaller strip
```
The example needs to be updated is all.

FlightDataRecorder, 2024-07-20:
Makuna;

    Thanks so much!  That was it!  As simple as that change was, not being familiar whit the project I had absolutely no idea where to start to figure that out by my self.

    I almost never support things like this, but I have also never got personal overnight service like that; so I made a donation to support this project.

    Also, are there any issues running NeoPixelBus on the 2nd Core of an RP2040? (i.e. Setup1 and Loop1.)

    Thanks again!
    FlightDataRecorder

Makuna, 2024-07-21:
Also, all the examples will work with RP2040 (should).  This sepecific one demonstrates the longer names so you can use different PIO hardware if you need.

The default method names like NeoWs2812xMethod are just an alias for respective names like Rp2040x4Pio1Ws2812xMethod.  So, this will work also...
```
NeoPixelBus<NeoBgrFeature, NeoWs2811Method> strip1(120, 15); // note: older WS2811 and longer strip
NeoPixelBus<NeoGrbFeature, NeoWs2812xMethod> strip2(100, 2); // note: modern WS2812 with letter like WS2812b
NeoPixelBus<NeoGrbFeature, NeoWs2812xInvertedMethod> strip3(100, 4); // note: inverted
NeoPixelBus<NeoGrbwFeature, NeoSk6812Method> strip4(50, 16); // note: RGBW and Sk6812 and smaller strip
```

Makuna, 2025-05-02:
https://github.com/Makuna/NeoPixelBus/releases/tag/2.8.4

closed 2025-05-02 by Makuna