Makuna/NeoPixelBus #817: UCS8903 & UCS8904 don't have name specific method

Makuna, 2024-07-20:
**Is your feature request related to a problem? Please describe.**
While the UCS890x chips do have specifically named features, they do not have specifically named methods.

**Describe the solution you'd like**
Methods should be aliased to either Ws2812x (most compatible) or another than has the nearest reset timing.

dimitre, 2024-08-02:
I'm trying to control an individual UCS8904B using NeoPixelBus without success.
I'm using a Teensy 3.2 and a simple code to make it blink. 
It behaves as it has no data incoming, only blue channel on, and the others off.
Thanks for any insight

```c++
#include <NeoPixelBus.h>
const uint16_t PixelCount = 10;
const uint8_t PixelPin = 11;
NeoPixelBus<NeoRgbw64Feature, Neo800KbpsMethod> strip(PixelCount, PixelPin);

void setup() {
	strip.Begin();
	pinMode(13, OUTPUT);
}

void loop() {
	Rgbw64Color cor(Rgbw64Color::Max, Rgbw64Color::Max, Rgbw64Color::Max, Rgbw64Color::Max);
	for (int a=0; a<NUMPIXELS; a++) {
		strip.SetPixelColor(a, cor);
	}
	strip.Show();
	digitalWrite(13, HIGH);

	delay(500);

	Rgbw64Color cor2(0, 0, 0, 0);
	for (int a=0; a<NUMPIXELS; a++) {
		strip.SetPixelColor(a, cor2);
	}
	strip.Show();
	digitalWrite(13, LOW);

	delay(500);
}
```

Makuna, 2024-08-02:
Try using "NeoWs2812xMethod"

dimitre, 2024-08-02:
Thanks, I've tried and it didn't work

dimitre, 2024-08-02:
from what I'm inspected here this NeoWs2812xMethod derives from NeoArmMk26z64SpeedWs2812x
which has the name similar to the Teensy LC instead of 3.2
Maybe I should test with another processor here.

Makuna, 2024-08-02:
Try the "NeoWs2805Method".

These ARM chips are painful to support.  Its bitbang model (no hardware support) and the uC speed variances can affect the output.  I would have to put it on the scope and I don't have my Arm chips with me to check it out right now.

Makuna, 2024-08-02:
In your compiler output, look for one of these...
__MK20DX128__, __MK20DX256__, __MK64FX512__, or __MK66FX1M0__

The LC will have __MKL26Z64__

dimitre, 2024-08-03:
Great. None of them worked.
I kept testing and reading code here, it seems to be everything allright with the timing, if I use a ws2812 strip it is working great. I have a chip holder and if I use the chip UCS2903 which uses the same protocol as ws2812 it works great also, but with UCS8904B it appears as no incoming data (only blue channel on)

I'm sharing the data pin with an WS2812 strip and it is blinking correctly, even when using 16bit code.
I'm wondering now if startFrame and endFrame packet of both protocols are equal, since I've used another strip (HD108S) and it needed a different number of bytes there.

Makuna, 2024-08-03:
NeoRgbw64Feature is for any chip that has 16bits per element with R,G,B, and a W channel.  This will not work with a WS2812 as it is only 8 bits per color element.  You would have to change it to NeoRgbwFeature for it to work with a WS2812.

dimitre, 2024-09-23:
Everything is now solved. 
The real issue was I was using a bidirectional level shifter and it was causing issues with the data.
I'm now  using 74AHCT125 and it is working great with UCS8904B