Makuna/NeoPixelBus #826: Teensy - missing uint8_t in latest

dimitre, 2024-08-24:
I'm having this issue when using latest from github
installing latest release in platformio it compiles OK
Thank you for the great lib!

```
.pio/libdeps/development/NeoPixelBus/src/internal/methods/FractionClk.c:40:32: error: unknown type name 'uint8_t'
   40 | void UnitDecimalToFractionClks(uint8_t* resultN,
      |                                ^~~~~~~
.pio/libdeps/development/NeoPixelBus/src/internal/methods/FractionClk.c:30:1: note: 'uint8_t' is defined in header '<stdint.h>'; did you forget to '#include <stdint.h>'?

```

Makuna, 2024-08-25:
thanks, it looks like on that platform stdio.h doesn't include stdint.h.  I will get a fix soon.

Makuna, 2024-08-26:
https://github.com/Makuna/NeoPixelBus/pull/827

Makuna, 2024-09-02:
https://github.com/Makuna/NeoPixelBus/releases/tag/2.8.1

closed 2024-09-02 by Makuna