Makuna/NeoPixelBus #843: Arduino Uno not working after version 2.8.0

dimitre, 2024-09-20:
If I use latest on Arduino IDE it doesn't compile and the error is this 
```
In file included from /Users/d/Documents/Arduino/libraries/NeoPixelBus_by_Makuna/src/NeoPixelAnimator.h:30:0,
                 from /Users/d/Documents/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/animations/NeoPixelAnimator.cpp:28:
/Users/d/Documents/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/animations/NeoEase.h:37:10: fatal error: functional: No such file or directory
 #include <functional>
          ^~~~~~~~~~~~
compilation terminated.
exit status 1
```

works ok if I revert to 2.8.0

Using platformio it doesn't compile, but it returns other error:
```
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/uno.html
PLATFORM: Atmel AVR (5.0.0) > Arduino Uno
HARDWARE: ATMEGA328P 16MHz, 2KB RAM, 31.50KB Flash
DEBUG: Current (avr-stub) External (avr-stub, simavr)
PACKAGES: 
 - tool-avrdude @ 1.60300.200527 (6.3.0) 
 - toolchain-atmelavr @ 1.70300.191015 (7.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/uno/src/main.o
src/main.cpp:6:10: fatal error: NeoPixelBus.h: No such file or directory
```

Makuna, 2024-09-21:
https://github.com/Makuna/NeoPixelBus/pull/844

Makuna, 2024-09-27:
https://github.com/Makuna/NeoPixelBus/releases/tag/2.8.3

closed 2024-09-27 by Makuna