Makuna/NeoPixelBus #858: Add support for GW6205 chipset

jerano, 2024-11-04:
I'd like to be able to use WLED, but my particular LED strip is currently not supported... and they kind of said if it's supported by NeoPixelBus it can be supported by WLED - So here by kindly ask if it's possible to add GW6205 support!?

GW6205 is supported by FastLED so hopefully it's not a super advanced thing to add support to NeoPixelBus?
I can provide logs from logic analyzer or similiar if wanted/required.

Thanks!

Makuna, 2024-11-04:
Find and reference the data sheet for the GW6205 is a start.  A quick searched came up with just references to FastLED.

jerano, 2024-11-04:
Trust me, I've tried... searched every corner of the vast internets to no avail!
But basically it's the same as the WS28xx chipset but with a 12b color depth.

12 bits / color @ 800kHz (or 400kHz)

Below you can have a look at the pulse train for 1 pixel to display RED.
[image]

Makuna, 2024-11-04:
Currently, non-byte aligned pixels is problematic with some of the methods.  There is a new branch, CoreShaderBeta that makes this easier and it's the future; but is far from ready as it has some major changes.

Patrick-Ohlson, 2024-11-11:
I would also love this implemented

Makuna, 2024-11-11:
Is it just 12 bits per color element?  Or does it include "padding bits" so that a single pixel of color is byte aligned?  Like 4.12.12.12, with four leading/trailing bits used for something else?

jerano, 2024-11-11:
As far as I can see, it's only 12+12+12 bits - see attached image on my previous post for a complete bitstream for one pixel.