Makuna/NeoPixelBus #863: Wrong implementation of "SinusoidalCenter"

SamFisher940425, 2024-11-29:
**Describe the bug**
There is an error in the function implementation of "SinusoidalCenter" that caused the output to be abnormal when the input was greater than 0.5

**To Reproduce**
Steps to reproduce the behavior:
In the file "src/internal/animations/NeoEase.h", line 214. In the function "SinusoidalCenter", I think there is an extra parenthesis in the 
code "return (-0.5f * (cos(PI * (unitValue-0.5f)) + 1.0f));", I think the  correct one is "return (-0.5f * cos(PI * (unitValue - 0.5f)) + 1.0f);"

**Expected behavior**
I think the  correct one is "return (-0.5f * cos(PI * (unitValue - 0.5f)) + 1.0f);", according to the picture "sinusoidal.png"

**Development environment (please complete the following information):**
 - Library version V2.8.0

SamFisher940425, 2024-12-03:
[image]
I've created a scatterplot using excel to illustrate the problem, and I hope it helps

SamFisher940425, 2024-12-06:
Thank you for your quick response and fixing!

Makuna, 2024-12-06:
I leave issues open and labeled "pending" until I make a release with the changes in it.

Makuna, 2025-05-02:
https://github.com/Makuna/NeoPixelBus/releases/tag/2.8.4

closed 2025-05-02 by Makuna