Makuna/NeoPixelBus #870: Add a method to determine RAM requirements for a bus

blazoncek, 2025-01-12:
**Is your feature request related to a problem? Please describe.**
To prevent memory depletion (by over-allocating too many buses) it would be nice to have a method that returns total amount of RAM needed for particular bus (including settings and backend buffers).

**Describe the solution you'd like**
At WLED we do not create a bus during startup like in most examples but instead use dynamically created buses using `new` operator. This means we can continue creating buses at runtime until memory runs out, if we do not track memory usage. Some buses use less RAM than others due to differences in underlying technology (RMT, I2S/DMA, etc) so it is challenging to correctly track RAM use.

It would be nice if we could get information how much total RAM will be used by particular bus *before* such bus is created.

Something like: `NeoPixelBus::GetTotalBuffers()` or `NeoPixelBus::GetTotalSize()` similar to `NeoPixelBus::PixelsSize()`.

**Describe alternatives you've considered**
We manually track and calculate (empirically) amount of RAM used.

**Additional context**
Sometimes underlying buffers change (i.e. 3/4 cadence with I2S) which will skew uneducated (empirical) manual tracking.

As some buses share same underlying buffers (i.e. parallel I2S) it is up to the calling application to take care of that.

Makuna, 2025-01-12:
You might also use it to test if a setup will fit in memory before instancing a bus object.  A static call that takes the pixel  count might help then.

The issue comes from the parallel busses.  Where front buffers are managed by instanced object and the back buffer is managed by a static shared objects.

blazoncek, 2025-01-12:
Yes, parallel busses may be challenging as they share some buffers but that could be taken into account when specifying length of 0.
In such case you could report only the size of existing back buffers (0 if there are none allocated).

My experimentation led me to this snippet:

```c++
unsigned BusDigital::getBusSize(unsigned count, uint8_t pin, unsigned type, unsigned nr) {
  byte pins[2] = {pin,0};
  unsigned iType = PolyBus::getI(type, pins, nr);
  return sizeof(BusDigital) + PolyBus::getBusSize(count, iType);
}

static unsigned PolyBus::getBusSize(unsigned count, unsigned busType) {
  unsigned size = 0;
  switch (busType) {
    case I_NONE: break;
...
    case I_32_RN_NEO_3: size = B_32_RN_NEO_3::GetBusSize(count); break;
    case I_32_I2_NEO_3: size = (_useParallelI2S) ? B_32_IP_NEO_3::GetBusSize(count)-B_32_IP_NEO_3::GetBusSize(0) : B_32_I2_NEO_3::GetBusSize(count)-B_32_I2_NEO_3::GetBusSize(0); break;
...
    }
    return size;
  }
```

`I_32_RN_NEO_3` is a unique NPB type and `B_32_RN_NEO_3` is `#define` for actual `NeoPixelBusLg<Color,Method,Gamma>` used