Makuna/NeoPixelBus #871: Official Support for UCS2904 NeoPixels

gcdinsmore, 2025-01-12:
A lot of NeoPixel pucks that are available are using the UCS2904 chips.  These are RGBW, with 8 bit dimming per channel.  It's looking like these are similar to the UCS8904 chips that are already supported by NeoPixelBus.  The difference is, UCS8904 is 16 bit dimmable per channel.

Source: https://www.artleds.com/blog/ucs8904-ic-pixel-protocol-overview

I have some of these UCS2904 chips.  I'm running them with WLED, and they seem to be working fine.  I'm sure you're aware, WLED uses NeoPixelBus for the interface to the NeoPixels.

What is required to officially support a chip / protocol?  I would be willing to execute what ever test is needed.

Makuna, 2025-01-12:
https://www.artleds.com/images/Specs/UCS2904-Datasheet.pdf
The data sheet doesn't call out what order or the timing for the protocol.  Since you are running them fine they are more than likely fully WS2812x compatible.  Are you selecting RGBW color order in WLED?

In NPB, the UCS8904 is just a "feature" selection, so color order and color bits.  Adding one specific to UCS2904 should be just as simple.  It just aliases the generic RGBW feature.

Now, the next issue is getting WLED to expose the name.  It will just wrap the current generic RGBW feature.  And this could be done within WLED without a change here.  But I will add the wrapper since I have one for the sister chip UCS8904.

gcdinsmore, 2025-01-12:
In WLED, I configured UCS2904 like this:

Type: SK6812/WS2814 RGBW
mA/LED: 30mA (typ. 12V)
Color Order: RGB 
Auto-calculate W channel from RGB: Accurate

gcdinsmore, 2025-01-12:
if you add the wrapper, I'll open a PR for WLED.  It'll be dependent on what ever version of NPB you release under, of course.  Thanks!

gcdinsmore, 2025-01-15:
Pure NPB code.  It works great with my UCS2904 pixels.

```
#include <NeoPixelBus.h>

const uint16_t PixelCount = 5;
const uint8_t PixelPin = 1;

#define colorSaturation 128

NeoPixelBus<NeoRgbwFeature, NeoWs2812xMethod> strip(PixelCount, PixelPin);

RgbwColor red(colorSaturation, 0, 0);
RgbwColor green(0, colorSaturation, 0);
RgbwColor blue(0, 0, colorSaturation);
RgbwColor white(colorSaturation);
RgbwColor black(0);

void loop()
{
  delay(5000);

  Serial.println("Colors R, G, B, W...");

  strip.SetPixelColor(0, red);
  strip.SetPixelColor(1, green);
  strip.SetPixelColor(2, blue);
  strip.SetPixelColor(3, white);
  strip.SetPixelColor(3, RgbwColor(colorSaturation));
  strip.Show();

  delay(5000);

  Serial.println("Off ...");

  // turn off the pixels
  strip.SetPixelColor(0, black);
  strip.SetPixelColor(1, black);
  strip.SetPixelColor(2, black);
  strip.SetPixelColor(3, black);
  strip.Show();
}

void setup()
{
  Serial.begin(115200);
  while (!Serial)
    ;  // wait for serial attach

  Serial.println();
  Serial.println("Initializing...");
  Serial.flush();

  strip.Begin();
  strip.Show();

  Serial.println();
  Serial.println("Running...");
}
```

rivernate, 2025-11-27:
Does this data sheet provide the missing information for the UCS2904? https://www.gree-leds.com/web/userfiles/download/UCS2904BGreeledEnglish.pdf