Makuna/NeoPixelBus #882: ESP32 DMA SPI examples do not work (DAT pin misused)

ShawnHymel, 2025-03-05:
In the ESP32 DMA SPI examples, you have the following (if SPI is configured):

```
strip.Begin(DotClockPin, DotDataPin, DotDataPin, DotChipSelectPin);
```

The issue is that `DotDataPin` is used twice: once for MISO and once for MOSI. Somewhere, it's set as both an input and output pin, and, as a result, the demo does not work. I recommend using a dummy pin or the default MISO pin (e.g. 19) for the second parameter to get it to work. 

I'm not sure where else this occurs in the library, hence the issue instead of a PR. This affects at least https://github.com/Makuna/NeoPixelBus/blob/master/examples/ESP32/DotStarTest_Esp32DmaSpi/DotStarTest_Esp32DmaSpi.ino

Makuna, 2025-03-05:
(this was a contributed example)

Did you try just passing -1 for the pin?  In general, on ESP32, using a pin of -1 means ignore the pin but I am unsure if that works for a possibly critical pin like MISO and MOSI.

ShawnHymel, 2025-03-05:
Yes, I can confirm that passing -1 for the MISO pin does work.

Makuna, 2025-03-05:
https://github.com/Makuna/NeoPixelBus/pull/883

Makuna, 2025-05-02:
https://github.com/Makuna/NeoPixelBus/releases/tag/2.8.4

closed 2025-05-02 by Makuna