Makuna/NeoPixelBus #886: ESP32 H2 not compiling with NeoPixelBus

fyrebug32, 2025-04-29:
I have used this amazing library with other ESP32 chips without issue but the newer bluetooth only H2 seems to be having issues with it.
This is the error copied from NeoPixeltest example, but it happens on all the examples. To reproduce the error using Arduino IDE 2.3.6 Just set the board to 'ESP32H2 Dev Module' and the compile will fail.

C:\Users\xx\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src\internal\methods\NeoEspBitBangMethod.cpp: In function 'bool neoEspBitBangWriteSpacingPixels(const uint8_t*, const uint8_t*, uint8_t, uint32_t, uint32_t, uint32_t, size_t, uint32_t, bool)':
C:\Users\xx\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src\internal\methods\NeoEspBitBangMethod.cpp:126:29: error: 'struct gpio_dev_t' has no member named 'out1_w1ts'; did you mean 'out_w1ts'?
  126 |         setRegister = &GPIO.out1_w1ts.val;
      |                             ^~~~~~~~~
      |                             out_w1ts
C:\Users\xx\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src\internal\methods\NeoEspBitBangMethod.cpp:127:31: error: 'struct gpio_dev_t' has no member named 'out1_w1tc'; did you mean 'out_w1tc'?
  127 |         clearRegister = &GPIO.out1_w1tc.val;
      |                               ^~~~~~~~~
      |                               out_w1tc
exit status 1

Compilation error: exit status 1

Makuna, 2025-04-29:
Core3 branch had this already fixed.  I merged the fix into master from Core3.

https://github.com/Makuna/NeoPixelBus/pull/887

Makuna, 2025-04-29:
let me know if this all works for you, then I will kick out a long due new release.

fyrebug32, 2025-05-01:
Yes! Thank you Makuna. I've tested multiple examples now and they are working with the H2 with the update!

Makuna, 2025-05-02:
https://github.com/Makuna/NeoPixelBus/releases/tag/2.8.4

closed 2025-05-02 by Makuna