Makuna/NeoPixelBus #890: Add support for CCT IC LED Light Strip WS2811 (NeoPixelBus)

Grappa2020, 2025-05-12:
please add support for CCT IC LED.

https://www.aliexpress.com/item/1005008771952476.html?spm=a2g0o.order_list.order_list_main.21.447e1802IhXKBX

This use the first two channels of WS2811 as Warm white (2700K) and Cold white (6500K) the third channel is un used.

suggesting to add The type of light (type: GRB) new options like CWN (Cold/Warm/Non) and any combinations.
i believe this mainly reflect on the Home Assistant GUI but very helpful when choosing white color temperature.

another option is to reflect the 3 or 4 channels of WS2811 as 3 or 4 separate light devices 

currently i can use this led strip but the way to control the white color temperature is mixing between red and green color 

thanks

Makuna, 2025-05-12:
**note**: while we all think LEDs are exactly 2700K or 5000K, some have values mapped to nearby frequencies but still call them Cold or Warm. So, any special methods to blend/calculate would be off from one LED to the other.  I avoid this whole issue in this library by using the terms Warmer and Colder, meaning relationships between the color channels but not being specific to a frequency range.

Are you asking for a new color object, RgbColor replacement that is white only?  
* CwColor (two channel, Colder and Warmer) 
* CnwColor (three channel, Colder, Neutral, Warmer)

 
Or are you asking for a new feature that maps the color object into the channels?
* NeoCwxFeature - (three channel, Colder, Warmer, and ignored)

Or both?

Short term, with the RgbColor object, you can reference the channels by index and ignore the R,G,B members.  The following examples consider the order as Colder, Neutral, Warmer.  I can see value in having a CnwColor and a CwColor object though for clarity.

```
// initialize 
RgbColor warmWhite = {0,0,255}; // colder, neutral, warmer
RgbColor coldWhite;

// index member access
coldWhite[0] = 255; // colder channel
coldWhite[1] = 0;  // neutral channel
coldWhite[2] = 0; // warmer channel
``` 

Then you can define and use your own color feature until it gets added.  Here is the implementation for a NeoCwxFeature that uses the RgbColor object (8 bits per color channel, in order of Colder, Neutral, Warmer). 

```
class NeoCwxFeature :
    public Neo3ByteFeature<0, 2, 1>,
    public NeoElementsNoSettings
{
};
```
Currently I don't have a Neo3xByteFeature but that can be added easily enough so it completely ignores the last channel.  But if the strip truly ignores the extra channel, it doesn't hurt that the above is setting it to a "second" index of the color let alone any index.

Grappa2020, 2025-05-12:
Sorry, i was not clear in my explanation.

i mention the color temperature just to be clear about Warm white/Cold white and i totally agree with you about using the terms Warmer and Colder, for home lighting the accurate color temperature is not important.

i was asking for new color object "CwColor (two channel, Colder and Warmer)" on WS2811 output 1 and 2.
but since i believe they can be many version of using WS2811 in the further i recommended to have the ability to map any color to any WS2811 output Red, Blue, Green, Colder, Warmer and N (Non) for no mapping.

i'm no familiar with RgbColor object and how to use it can you please direct me to the documentation about it.

the strip that i have is ignoring the 3rd output, its not connected to anything.

Makuna, 2025-05-14:
https://github.com/Makuna/NeoPixelBus/wiki/RgbColor-object-API